Protonerd/FX-SaberOS #1: sound and color

DRagonenko, 2017-04-26:
https://youtu.be/4xO36PvQsWU
correctly plays audio files . the blade color is not changing =(
Sorry if I have mistakes. I'm not good enough in English

jbkuma, 2017-04-26:
The color change is the "Gravity color" from LSOS 1.5

Here is a video demonstrating the color:
https://www.youtube.com/watch?v=crkaLPb7E7s

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

DRagonenko, 2017-04-26:
sound problems 
Problems with the changing color of the blade
https://youtu.be/3OcxuN5DtIA

jbkuma, 2017-04-26:
https://youtu.be/3OcxuN5DtIA?t=75

The part that is red is the color to be added or subtracted.  You must turn
your board to different angles to make this bright to add, or dim to
subtract.  I can see when you turn the board the end changes color
correctly.  The button must be pressed when the tip is on the correct color.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

DRagonenko, 2017-04-26:
does not retain changes
https://youtu.be/tnWBY4ims0E

DRagonenko, 2017-05-15:
not plays those files

closed 2017-11-04 by Protonerd