Protonerd/FX-SaberOS #3: info

shiryu55, 2017-06-22:
estoy mirando hace 4 días para crear unas bases de sonido y tengo una serie de preguntas rápidas.

Mis dudas son:

son necesarios 4 ficheros de arranque y 2 de power off.?? o solo con uno para cada evento es suficiente.

Do we need 4 "Start Files" and 2 "power off files"? Or is it enough with just one for each session
--------------------------------------------------------
los efectos del swing , estan numerados del 008- al 015 en la base original 1,5, unos son sonidos rapidos otros mas lentos,, 

The effects of SWING are numered from 008-015 on the original base. 1'5,. Some sound fast and others slow.

 cuando genere con el programa otra fuente de sonido, como tengo que ordenar los sonidos del swing??

How do I order the sound of the swing when I generate with the programme another source of sound?

- en funcion de la rapidez del efecto??, o mejor dicho, como sabe el programa que efecto de sonido tomar en función de la velocidad de movimiento..... o pilla un sonido aleatorio de entre los ficheros designados para el swing.

What about the speed? Or better still, how does the program know what sound effect to choose from according to the movement. Or is it just a random sound amongst all the files designated for the "swing"

--------------------------------------------------------

segun la programacion del fichero sonund.h. y la base de sonidos de la version 1.5

According to the file sound.h program and the list of sounds, version 1.5

son 1 boot, 4 power on, 2 power off, 8 swing, 8 clash, 1 locupk, 4 blaster, 1 menu, 1 hum

They are:
1 boot, 4 power on, 2 power off, 8 swing, 8 clash, 1 locupk, 4 blaster, 1 menu, 1 hum

en el programa de jackesoft,  seria algo asi ??

would it be the following for the Jackesoft Program?

sound counts:
power up: 4
power down:2
swing:8
clash:8
blaster:4
lockup:1
boot:1
spin:--- don't know what files
stab:------ don't know what files
force:-----don't know what files

source- .. understood
destination. Understood

tarjet lenght: 120s

pinchar en convert-
Click on Concert-

estas son las fuentes de sonido. a utilizar. y el tutorial del programa que indicais .

These are the sources of sound to use and the tutorial of the program you provide.

http://www.saberforum.com/index.php?topic=33475.0

https://www.youtube.com/watch?v=8rtOaMDS7Lw

jbkuma, 2017-06-22:
Soundfont.h can be changed to work with any set of sounds. I believe it is
configured for one on, one off. You need to look at the soundfont.h file.
Change it if you want to. The file is fully explained in LSOS GitHub and
won't be repeated here because it will change before it is released. We
won't be sharing any more information on that until it's ready.

 Nothing has changed when the on/off timing. It's the same settings we've
discussed before.

          this->powerOnTime = 300;
          this->powerOffTime = 700;

Sounds are randomly selected when an event occurs.  You can alter the
function if you want, but that's up to you.

http://www.saberforum.com/index.php?topic=33475.0

https://www.youtube.com/watch?v=8rtOaMDS7Lw

Protonerd, 2017-11-04:
Closing as no update for a long time.

closed 2017-11-04 by Protonerd