Protonerd/FX-SaberOS #4: Compilation Error Arduino Nano V3

Tam-Koowan, 2017-07-12:
Hello, I have no problem during the compilation but at the time of the upload I have a mistake.

Can you help me
I have an Arduino Navo v3 with LED Neopixel.

Here is the copy of the errors:

Arduino : 1.6.11 (Windows 7), Carte : "Arduino Nano, ATmega328"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Chris\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Chris\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10611 -build-path C:\Users\Chris\AppData\Local\Temp\build9f79cc7659e0d95d793de6397bf3f02b.tmp -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -verbose C:\Users\Chris\AppData\Local\Temp\arduino_modified_sketch_605465\FX-SaberOS.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Chris\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Chris\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10611 -build-path C:\Users\Chris\AppData\Local\Temp\build9f79cc7659e0d95d793de6397bf3f02b.tmp -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -verbose C:\Users\Chris\AppData\Local\Temp\arduino_modified_sketch_605465\FX-SaberOS.ino
Using board 'nano' from platform in folder: C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19
Using core 'arduino' from platform in folder: C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19
Detecting libraries used...
"C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\eightanaloginputs" "C:\Users\Chris\AppData\Local\Temp\build9f79cc7659e0d95d793de6397bf3f02b.tmp\sketch\FX-SaberOS.ino.cpp" -o "nul"
"C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\libraries\DFPlayer_LSOS" "C:\Users\Chris\AppData\Local\Temp\build9f79cc7659e0d95d793de6397bf3f02b.tmp\sketch\FX-SaberOS.ino.cpp" -o "nul"
"C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\libraries\DFPlayer_LSOS" "-IC:\Program Files (x86)\Arduino\libraries\I2Cdev" "C:\Users\Chris\AppData\Local\Temp\build9f79cc7659e0d95d793de6397bf3f02b.tmp\sketch\FX-SaberOS.ino.cpp" -o "nul"
"C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\libraries\DFPlayer_LSOS" "-IC:\Program Files (x86)\Arduino\libraries\I2Cdev" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src" "C:\Users\Chris\AppData\Local\Temp\build9f79cc7659e0d95d793de6397bf3f02b.tmp\sketch\FX-SaberOS.ino.cpp" -o "nul"
"C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\libraries\DFPlayer_LSOS" "-IC:\Program Files (x86)\Arduino\libraries\I2Cdev" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\MPU6050" "C:\Users\Chris\AppData\Local\Temp\build9f79cc7659e0d95d793de6397bf3f02b.tmp\sketch\FX-SaberOS.ino.cpp" -o "nul"
"C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\libraries\DFPlayer_LSOS" "-IC:\Program Files (x86)\Arduino\libraries\I2Cdev" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\MPU6050" "-IC:\Program Files (x86)\Arduino\libraries\EEPROMEx" "C:\Users\Chris\AppData\Local\Temp\build9f79cc7659e0d95d793de6397bf3f02b.tmp\sketch\FX-SaberOS.ino.cpp" -o "nul"
"C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\variants\eightanaloginputs" "-IC:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\MPU6050" "-IC:\Program Files (x86)\Arduino\libraries\EEPROMEx" "-IC:\Program Files (x86)\Arduino\libraries\OneButton" "-IC:\Program Files (x86)\Arduino\libraries\DFPlayer_LSOS" "-IC:\Program Files (x86)\Arduino [... cut ...]
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\_2_Servo_ACC : D:\Documents\Arduino\libraries\_2_Servo_ACC
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\MMA8451-master : D:\Documents\Arduino\libraries\MMA8451-master
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\Servo_2_accelerometre : D:\Documents\Arduino\libraries\Servo_2_accelerometre
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\_2_Servo_ACC : D:\Documents\Arduino\libraries\_2_Servo_ACC

Protonerd, 2017-07-13:
Can you please switch off verbose, it's impossible to find the real error amidst all the text.

Tam-Koowan, 2017-07-13:
J'ai enfin réussi à téléversé, mais ça ne fonctionne pas  :( 
Il y a une manipulation à faire pour la première utilisation ?

DRagonenko, 2017-07-17:
библиотеки установи нормально

Protonerd, 2017-11-04:
Closing as no update for a long time.

closed 2017-11-04 by Protonerd

Tam-Koowan, 2017-11-05:
Thank you for closing!
Nobody helps so I have nothing to say that's why I did not answer!

Vulkan1, 2017-11-05:
Gam you need to turn off Verbose, otherwise they can’t help you. What you sent them was a needle in a hay stack. They were waiting for your response with the new problem log.

Tam-Koowan, 2017-11-05:
I'm fine but I make a comment in the setting?

Tam-Koowan, 2017-11-05:
I check only "default" and I have this as a mistake?

Arduino : 1.6.8 (Windows 7), Carte : "Arduino Leonardo"

In file included from D:\Documents\Arduino\FX-SaberOS\FX-SaberOS.ino:37:0:

sketch\Config.h:235:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\Config.h:258:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST 

       ^

sketch\Config.h:273:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

D:\Documents\Arduino\FX-SaberOS\FX-SaberOS.ino:122:7: warning: extra tokens at end of #else directive

 #else if #ifdef DIYINO_STARDUST

       ^

D:\Documents\Arduino\FX-SaberOS\FX-SaberOS.ino:1036:13: warning: extra tokens at end of #else directive

       #else if STAR_LED

             ^

FX-SaberOS:23: error: 'DFPLAYER_TX' was not declared in this scope

 SoftwareSerial mp3player(DFPLAYER_TX, DFPLAYER_RX); // TX, RX

                          ^

FX-SaberOS:23: error: 'DFPLAYER_RX' was not declared in this scope

 SoftwareSerial mp3player(DFPLAYER_TX, DFPLAYER_RX); // TX, RX

                                       ^

In file included from D:\Documents\Arduino\FX-SaberOS\FX-SaberOS.ino:28:0:

DFPlayer.h:133: error: 'DFPlayerSerial' does not name a type

  inline DFPlayerSerial * getSerial() {

         ^

DFPlayer.h:626: error: 'DFPlayerSerial' does not name a type

  DFPlayerSerial * mp3Serial;

  ^

D:\Documents\Arduino\libraries\DFPlayer/DFPlayer.h: In destructor 'DFPlayer::~DFPlayer()':

DFPlayer.h:128: error: 'mp3Serial' was not declared in this scope

   if (NULL != mp3Serial) {

               ^

[... 250 lines cut ...]
                                                ^

C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino/Arduino.h:123:25: note: in definition of macro 'bit'

 #define bit(b) (1UL << (b))

                         ^

Plusieurs bibliothèque trouvées pour "DFPlayer.h"
Utilisé : D:\Documents\Arduino\libraries\DFPlayer
Non utilisé : D:\Documents\Arduino\libraries\LightSaberOS
Non utilisé : D:\Documents\Arduino\libraries\FX-SaberOS
Non utilisé : D:\Documents\Arduino\libraries\DFPlayer_LSOS
Plusieurs bibliothèque trouvées pour "Wire.h"
Utilisé : D:\Documents\Arduino\libraries\Wire
Non utilisé : C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\Wire
Plusieurs bibliothèque trouvées pour "LinkedList.h"
Utilisé : D:\Documents\Arduino\libraries\LinkedList
Non utilisé : D:\Documents\Arduino\libraries\FX-SaberOS
exit status 1
'DFPLAYER_TX' was not declared in this scope

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.