Protonerd/FX-SaberOS #5: fail compilation arduino ide 1.6.11. avr 1.6.18  arduino nano v3.

shiryu55, 2017-09-16:
class dfplayer has no member named sleep.

---------

Arduino:1.6.11 (Windows 10), Tarjeta:"Arduino Nano, ATmega328"

In file included from sketch\Buttons.cpp:10:0:

sketch\Config_HW.h:46:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\Config_HW.h:77:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST 

       ^

In file included from sketch\Buttons.cpp:10:0:

sketch\Config_HW.h:97:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\Config_HW.h:125:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

In file included from sketch\Buttons.cpp:12:0:

sketch\SoundFont.h:23:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\SoundFont.h:96:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\SoundFont.h:135:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST // 18 files per sound font

         ^

In file included from sketch\ConfigMenu.h:9:0,

                 from sketch\ConfigMenu.cpp:8:

sketch\Config_HW.h:46:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\Config_HW.h:77:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST 

       ^

In file included from sketch\ConfigMenu.h:9:0,

                 from sketch\ConfigMenu.cpp:8:

sketch\Config_HW.h:97:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\Config_HW.h:125:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

In file included from sketch\ConfigMenu.cpp:12:0:

sketch\SoundFont.h:23:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\SoundFont.h:96:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\SoundFont.h:135:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST // 18 files per sound font

         ^

sketch\ConfigMenu.cpp: In function 'void NextConfigState()':

sketch\ConfigMenu.cpp:96:24: warning: invalid conversion from 'int' to 'ConfigModeSubStatesEnum' [-fpermissive]

     ConfigModeSubStates=-1;

                        ^

sketch\ConfigMenu.cpp:98:44: warning: invalid conversion from 'int' to 'ConfigModeSubStatesEnum' [-fpermissive]

     ConfigModeSubStates=ConfigModeSubStates+1; // change to next config state in the ordered list

                                            ^

In file included from F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino:24:0:

sketch\Config_HW.h:46:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\Config_HW.h:77:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST 

       ^

In file included from F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino:24:0:

sketch\Config_HW.h:97:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\Config_HW.h:125:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

In file included from F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino:28:0:

sketch\SoundFont.h:23:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\SoundFont.h:96:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\SoundFont.h:135:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST // 18 files per sound font

         ^

F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino:111:9: warning: extra tokens at end of #else directive

   #else if #ifdef DIYINO_STARDUST

         ^

F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino:984:15: warning: extra tokens at end of #else directive

         #else if STAR_LED

               ^

F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino: In function 'void loop()':

F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino:946:26: warning: invalid conversion from 'int' to 'ConfigModeSubStatesEnum' [-fpermissive]

       ConfigModeSubStates=-1;

                          ^

F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino: In function 'void Set_Volume(int8_t)':

F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino:1324:36: warning: default argument given for parameter 1 of 'void Set_Volume(int8_t)' [-fpermissive]

 void Set_Volume(int8_t volumeSet=-1) {

                                    ^

F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino:57:6: note: previous specification in 'void Set_Volume(int8_t)' here

 void Set_Volume(int8_t volumeSet=-1);

      ^

F:\SABLE LASER\FX\FX-SaberOS\FX-SaberOS.ino: In function 'void SleepModeEntry()':

FX-SaberOS:1397: error: 'class DFPlayer' has no member named 'sleep'

   dfplayer.sleep();

            ^

Se encontraron múltiples librerías para "Wire.h"
Usado: C:\Users\fer2\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.18\libraries\Wire
 No usado: F:\SABLE LASER\arduino-1.6.11\libraries\Wire
Se encontraron múltiples librerías para "DFPlayer.h"
Usado: F:\SABLE LASER\arduino-1.6.11\libraries\DFPlayer
 No usado: F:\SABLE LASER\arduino-1.6.11\libraries\DFPlayer_LSOS
exit status 1
'class DFPlayer' has no member named 'sleep'

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

Protonerd, 2017-09-19:
did you do any change to the code?

shiryu55, 2017-09-20:
No, just download the arduino ide, copy the libraries on your site, select fx saberos.ino, compile

arduino nano v3, placa homebrew.

Protonerd, 2017-09-20:
Which IDE version are you using? And which Board manager version? Please have a look at https://github.com/Protonerd/DIYino/blob/master/DIYino_Stardust_v2_User_Manual_v1.pdf. Chapter 8 for a step by step walkthrough is there.

shiryu55, 2017-09-20:
ide 1.6.11. avr 1.6.18

shiryu55, 2017-09-20:
right now I just tried it on another pc,
download version ide 1.8.2
download fx-saberos from git protonerd
avr board 1.6.17

same failure,   class dfplayer has no member named sleep

jbkuma, 2017-09-20:
just comment out the line.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Protonerd, 2017-09-20:
Thanks Jason, yes, this line can be ommitted, it does not work anyway, the DFPlayer will refuse to go to sleep mode if it has any... actually I'm a bit surprised, I thought I removed this code a long time ago. I will check when I get home again.

Protonerd, 2017-09-20:
created a new commit where this function is not called.

Protonerd, 2017-11-04:
This bug is fixed, closing.

closed 2017-11-04 by Protonerd