Protonerd/FX-SaberOS #6: colors

DRagonenko, 2017-10-05:
have the ability to disable the gravity of the colors in the new code?

Protonerd, 2017-10-06:
I will not disable it, as I'm very fond of this method, currently it's the most sophisticated way to change color among all saber boards. But you can disable it by simply omitting the color change items from the config menu by setting them behind CS_LASTMEMBER. If you do so, you must define the color profiles in the code.

DRagonenko, 2017-10-08:
please make a screenshot where it is in the code.
thank you

Protonerd, 2017-11-04:
Do you need help here still? Let me know, otherwise close the topic.

DRagonenko, 2017-11-07:
I do not understand where is the code change.
please make a screenshot

Protonerd, 2017-11-14:
Have a look at the Wiki: https://github.com/Protonerd/FX-SaberOS/wiki/FlexiConfig

closed 2017-11-14 by Protonerd