Protonerd/FX-SaberOS #8: swingthreshold,powerOnEffect,flickerEffect

shiryu55, 2017-10-15:
in this new version, how to control the sensibility,

in the previous version lightsaber os you changed the parameters in this part of the code

/*** SOUNDFONT 2 ***/  
        case 3:           
          this->powerOnTime = 300;
          this->powerOffTime = 700;
          this->powerOnEffect = 0;
          this->powerOffEffect = 0;
          this->flickerEffect = 2;
          this->swingthreshold = 200;
          boot[0] = NR_CONFIGFOLDERFILES + NR_JUKEBOXSONGS + NR_SF1_FILES + 1;
          boot[1] = boot[0];
          powerOn[0] = boot[1] + 1;
          powerOn[1] = powerOn[0];
          powerOff[0] = powerOn[1] + 1;
          powerOff[1] = powerOff[0];
          swing[0] = powerOff[1] + 1;
          swing[1] = swing[0] + 7;
          clash[0] = swing[1] + 1;
          clash[1] = clash[0] + 7;
          lockup[0] = clash[1] + 1;
          lockup[1] = lockup[0];
          blaster[0] = lockup[1] + 1;
          blaster[1] = blaster[0] + 3;
          menu[0] = blaster[1] + 1;
          menu[1] = menu[0];
          hum[0] = menu[1] + 1;
          hum[1] = hum[0];
          break;

Protonerd, 2017-10-23:
should work the same way, except it might be defined somewhere else. Look for swing sensitivity. Alternatively the new release also has a config mode menu where you can change the sensitivity.

shiryu55, 2017-10-23:
Protonerd Thank you for answering so quickly

for example
in case 1 for example. mod pre establish,  powerOnEffect, powerOffEffect,swingthreshold
it is correct to configure it

or is modified in other parts of the code, to preset these effects of sensibility, flicker. etc..

#ifdef DIYINO_PRIME
    case 0:
      // soundFont directory 01 :
      this->powerOnTime = 800;
      this->powerOffTime = 800;
      break;
    case 1:     
      // soundFont directory 02 :
      this->powerOnTime = 1050;
      this->powerOffTime = 900;
      this->powerOnEffect = 0;  ------
     this->powerOffEffect = 0;--------
     this->flickerEffect = 2;--------
      this->swingthreshold = 200;-----
      break;

Protonerd, 2017-10-28:
Swing sensitivity/threshold is now a config variable stored in EEPROM, a global variable in the config stucture::
uint16_t swingSensitivity;
It defaults to 1000, which works remarkably well, if you want to change it, there is a dedicated config mode item CS_SWINGSENSITIVITY that can be used to set the sensitivity level on-the-fly.

Protonerd, 2017-11-04:
Did you try the swing sensitivity option? You can try the new commit, it contains some major bug fixes.

shiryu55, 2017-11-04:
I'm not at home to prove it. tomorrow I can try it

Protonerd, 2017-11-14:
Do you still need support here? Otherwise I would like to close.

shiryu55, 2017-11-14:
Close

closed 2017-11-20 by Protonerd