Protonerd/FX-SaberOS #9: Current draw? Single LiPo and RGB led

kirkyb, 2017-10-22:
I've recreated figure 13 from the Stardust user manual using +ve straight to blue and green led and resistor to red. The board functions perfectly fine with the RGB disconnected but crashes out when it is connected and 'ignited'.
I've tried to reduce the brightness in the Arduino code (i.e. to reduce current draw) but this doesn't appear to affect the RGB led brightness. Is there any other setting that can be altered?

jbkuma, 2017-10-22:
What size battery are you using? It probably shouldn't be drawing more than
3amps.  Are you using resistors?

kirkyb, 2017-10-22:
I'm running everything from a single 18650. Using 3.3ohm 2w resistor on red channel to drop from 3.7v to 2.8v (as specified) and running straight from +ve for blue and green

Protonerd, 2017-10-22:
What do you mean by +ve? I would connect one LED at a time and see which one causes the problem. Provided all the connections are right.
BTW, the Figure 13 I just realized can be misleading. The red connection which is suposed to show how to connect BAT+ to 5V might be interpreted as connecting all 4 signals together. I hope you did not do this.

Protonerd, 2017-10-22:
I will upload a new UM version

kirkyb, 2017-10-22:
Hi Andras, first of all thanks for the replacement board. I'm powering the blue and green led straight from positive of the battery and the red I'm using a 2w resistor in series. No, thankfully i realised that it is only the 2 signals and not all 4

Protonerd, 2017-10-22:
Worst case you bricked one of the transistors, that can happen quite easily if you make measurements while powering the LEDs and accidentally shorting an LED, which will send the whole current through the transistor, killing it. That is why I would go on the LS pins one by one seeing which is reacts strangely.

kirkyb, 2017-10-22:
R, G and B ask light individually without issue. Ignition seems to work fine but its the clash that actaully crashes the board. Just tried cycling through the menu and the board becomes unresponsive when it reaches clash colour.

Protonerd, 2017-10-22:
What type of battery are you using? The default for the color mix is white for a virgin board. Maybe not the best choice as it will active all LEDs and might trip the current protection of your battery. Do you have a protected battery?

kirkyb, 2017-10-22:
Its an ultrafire 18650 and yes its protected. Could i change the default color then if this is the possible issue?

Protonerd, 2017-10-23:
You can change in the FX-SaberOS.ino, line 197 change
if (!loadConfig()) {
to
if (true) {

and change the values in lines 209 - 217 all to 0.

Then try, if works, change back the line 197 to its original.

But they all have values of 20, a very low PWM value, so no real current should flow. I'm surprised. I would recommend to only connect one color at a time and try all functionalities. This way you might even change the colors in the config without changing the code.

kirkyb, 2017-10-24:
I've tried your suggestion but that didn't work.
But....
...tonight I used my 18650 battery for a pumpkin prop at my sons halloween party and upon plugging the battery back into the stardust I've noticed that in the menu it now says battery 'nominal' and not 'full'.
The system is much more stable, the LED illuminates with all blade functionality and i can cycle through the full menu without crashing.

Protonerd, 2017-10-28:
Hi Kirk, do I read it right, does the saber work now? I have no clue about the erronous behaviour with the fully charged battery to be honest, but if you have new findings to shed some light on it, let us know.

kirkyb, 2017-10-28:
Hi Andras,

The battery issue hasn't returned thankfully. A strange one really but i
guess it has something to do with its protection.
I swapped out the red LED resistor too for one with a higher power rating
and it was still giving me issues. I've since decided to omit this LED and
stick with only blue and green, it works perfectly. I can now increase
brightness to 255 for the remaining LEDs.

I aim to dismantle my stunt saber tonight and proceed with the install.
Have you any experience removing hot glue from the inside where the button
is?

Also, I've finally had the chance to try out the saber font that i
reproduced for the Stardust, it appears to work very well so far!

Kirk

Protonerd, 2017-10-30:
Glad to hear that. I still think that if the 3rd LED does not work it might be due to a bricked transistor.

kirkyb, 2017-11-06:
Just returning to this issue. A gut feeling told me to try using the v1.1 update with a second RGB LED that I owned (same supplier). Now I dont know if it was a faulty LED previously or whether its because of the new code, but I now have red working perfectly!

I have only had a sound distortion issue twice since installing the RGB but a quick charge overcame this.

I will monitor closely whilst draining some of the battery and will close the issue if all is well.

jbkuma, 2017-11-06:
Glad to hear you've got it sorted.

Sound distortion can be caused by a low battery for any saber board. The
sag effect, a voltage drop under load, can make this worse especially if
the battery isn't properly designed for high current.  The fact that
charging ameliorates the effect is the surest sign.

As to the LED issue, it's a head scratched based on your description. Could
be an odd issue or out of spec, it's hard to tell b from here.

Protonerd, 2017-11-14:
Closing as it looks to be solved.

closed 2017-11-14 by Protonerd