Protonerd/FX-SaberOS #10: Swing detection

Argathena, 2017-10-27:
I've got a 90% functional saber. The 10% missing is all swing. I'm hoping it's not obvious, but not fully knowing how to code, I can't figure it 

Clash works perfect along with color change. It just won't react on swing, at all.  

Using single button on star led in the config on a stardust v2

If someone could nudge me in the right direction and point out my obvious mistake. 

Ps I believe it's somehow in the code,  cause prior to getting my stardust, I was using same code configured for my Prime, and the swing would not react on it either.  Same set up there too, single button, star led, Diyino Prime.

Protonerd, 2017-10-28:
I have to look into it. Swing sensitivity defaults to 1000 upon first config and the only way to change it is to set it in the config menu CS_SWINGSENSITIVITY. If you have this config item, it could be that now you have a less sensitive setting which might make it difficult to make a swing. How about blaster deflect on the move? If you hit the button 2 times you arrive into the blaster deflect mode activated by gestures. Does that work? That one does not need wide sweeping gestures to be activated, a small but forceful jerk of your wrist if sufficient.

Argathena, 2017-11-04:
The Swing is still non functional for me.  The deflect does not work either. I tried the double click and it did not work with a full swing nor a quick jerk.   I have tried cycling through the sensitivity settings in the config mode and tried turning it up.  i have also tried just setting to 1000 and uploading, then testing.  Everything else works perfectly.  I have also re calibrated the MPU multiple times just in case.  

Sincerely
The Lost Jedi  ;-)

Protonerd, 2017-11-05:
If you omit the swing sensitivity from the config menu, sensitivity will default to 1000 and that works quite realiable. Amyway, sensitivity is a tricky beast. It goes from 0 - 16000 (roughly). But due to the current quaternion based implementation of the swing detection, it you increase sensitivity to above 3000, swings will be disabled. But jerks in blast deflect will still work. But will become increasingly difficult to trigger. If it does not, it might be that you need to exercise double click.

Why did you recalib the MPU? All Stardust boards come pre-calibrated... where did you get the calibration code? I have mine which writes the calibrated values to the right place, if you just download it from elsewhere it will not store the values.

One more thing: if you activate any of the fire blade effects in the flicker styles, swings are disabled. It is done on purpose.

One other thing:

Argathena, 2017-11-05:
I Re-calibrated the MPU just to try anything.  I used the sketch that came with FX-OS in the tools folder. It successfully calibrated and uploaded to the EEPROM. I knew they came pre-Calibrated but like i said ive been trying anything almost to figure the swing issue.  

**"One more thing: if you activate any of the fire blade effects in the flicker styles, swings are disabled. It is done on purpose."**   When you say this do you mean, when I choose from the flicker sytles in the config menu?  Or do you mean in the code?  I didnt see anywhere in the code that allows me to define fire blade AND Star Led.   Please clarify.

Protonerd, 2017-11-06:
Sorry, fireblade only exists of course for pixelblades. If you use star LED, you will not have this topic. For star LED you can still choose flicker styles, currently you can select between pulse mode, AudioTracker and static blade.

Protonerd, 2017-11-06:
Regarding the swing issue: try to play around with the swing sensitivity in config mode. In high sensitivity (red), even small movements will trigger a lot of repeated swings.

Argathena, 2017-11-06:
Still no swing at all, not even one.  Ive tried removing Swing Sensitivity from config menu so it defaults out.  Ive tried setting the sensitivity to High (Red) and Low (Green?) Ive tried the middle.  Ive tried the double click for Deflect and nothing.  This is where i really wish i knew C++ alot more.  

Silly question but does the Swing work on your Star LED with Stardust board?  If so is possible that you could upload that copy.  Maybe I changed something that i shouldnt have.  All I changed is Define Starled, Define Single Button, Define Blade Y and Changed the Max Brightness to 100.  Also arranged Config menu and put sensitivity at the front for easy testing.

Protonerd, 2017-11-07:
The code currently on GitHub (master branch) is exactly the code I upload to my sabers. I only change blade type to STAR_LED, board to STARDUST, and depending on the button setup SINGLEBUTTON on/off. Nothing more.

I conclude that it's not a problem of the sketch, nor of the boards (chance of 2 Stardust boards with defect MPUs which cannot be recognized by my test sketch is remote at best). It must be somewhere else, but we will find it.

I also looked at your debug log, it looks like you successfully activated motion triggered bolt deflect, which is good. I think there is no debug message on a swing, but one can be easily added, look for this:
        ActionModeSubStates = AS_SWING;
        SinglePlay_Sound(soundFont.getSwing((storage.soundFont)*NR_FILE_SF));
        /* NORMAL SWING */

Then insert afterwards:
Serial.println("Swing detected");

You shoud now see if a swing is triggered.

Could it be that you do not have swing sound? What did you upload as sound fonts?

Another reason can be still the MPU calibration. TO be on the safe side please rerun my outgoing test sketch on your Stardust boards:
https://github.com/Protonerd/DIYino/blob/master/DIYino_Testsketch_Pro.ino

...and send me the Serial Monitor log. When you see the calibrated MPU values, try to move the boards. For sure during the calibration process the board must stay motionless, facing upwards.

Argathena, 2017-11-08:
Well first i have been un-commenting 
**devStatus = mpu.dmpInitialize_light();  // this is a ligter version of the above**

It errors out each time, so i Omit.  I hope that it wasnt this the whole time..

The deflect is not working if thats what you mean from the log file.  
These two entries happen at the same time, before i try to swing.
**Start motion triggered blaster bolt deflect
Start motion triggered blaster bolt deflect**

I also added Serial.println("Swing detected");   Where you said too and it never showed in this test run.

The sound files that im using are the ones from the FX-saber OS/Soundfont/DIYino_SPIFlash_Image.7z 
I have made no changes to sound. 
 Im using 
SF01_VanillaLyte
SF03_VanillaSwirl
SF05_FallenSon

I ran the DIYino Test Sketch.  Board was motionless during upload and not moved until calibrated.  After that i swung it around as best i could while tethered.  Seen plenty of movement, or so it seemed to me.
Downloaded a fresh copy of the master, and tried again.  (Once again un-commenting devStatus = mpu.dmpInitialize_light();  ) Still nothing.  

I also tried to manual enter the offsets and un-commented **#define MPUCALOFFSETEEPROM** this also did not work.  i Turned off **#define SWING_QUATERNION** just to try, no swing.  Im trying anything 
[LS Debug 11-7-2017.txt](https://github.com/Protonerd/FX-SaberOS/files/1452262/LS.Debug.11-7-2017.txt)
[MPU Log 11-7-2017.txt](https://github.com/Protonerd/FX-SaberOS/files/1452263/MPU.Log.11-7-2017.txt)

lol.

Protonerd, 2017-11-08:
Arghhhhh, if you omit devStatus = mpu.dmpInitialize_light(); // this is a ligter version of the above, Swings will sure not work!!! All the MPU resources needed for the Quaternion based swing detections are in there.

The reason you have errors is that you have an outdated library. Please download all the libraries and overwrite the old ones you have.

closed 2017-11-08 by Protonerd

Argathena, 2017-11-08:
I knew this was my fault! Lol sorry for our headaches. Once again I know enough to mess it up. Thank you so much.  Works like a charm!!

Protonerd, 2017-11-08:
I'm glad it works now, if you have pics or a small video of your saber, share it!

Argathena, 2017-11-10:
I shall, some time shortly. I love the deflection!