Protonerd/FX-SaberOS #11: homebrew setup reboots/crashes when in the config menu

raptordemon, 2017-10-27:
Hi, first off, i love this project, it is awesome and thanks for building it.

i have build a homebrew version as i had all the parts in stock and decided to try it for halloween.

I've run into issues where the saber crashes while going through the config menu and it just reboots. ive also had issues with not being able to navigate the menus, usually around the Sleep mode section.

have you run into this before?

thanks
Raptor

Protonerd, 2017-10-28:
Could it be that you accidentally activate sleep mode? When sleep mode is entered, the board plays beep sound four times. Afterwards the controller enters sleep mode and can be woken up with pressing the main button. As to crashing in config mode, do you have lights? Did you check your voltages? Undervoltage can cause crash. As well as memory overflow if you use too many neopixels.

raptordemon, 2017-10-30:
it is possible that it is undervolting, i have been using an 18650 to power it directly. I'll install a boost converter and see if that helps.

for the Pixels i am using a strip of about 60. is that too many?

thanks
Raptor

jbkuma, 2017-10-30:
I don't recommend using a boost converter, it just eats more power
unnecessarily.  Check the quality and rating of your battery.  It needs to
be able to handle the current demand of the pixels.  It could be that the
battery's protection circuit is kicking in.  If you are in the US I
recommend Orbtronic, they have an 18650 rated at 10amps and 3500mah.

You can use up to 120 safely, the main issue is the memory requirements
involved.  We've been working on improving the efficiency of the code, and
you may be able to go higher, but that is our current recommendation.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

raptordemon, 2017-11-01:
ok this is power related. works fine if i plug it in! thanks everyone!

closed 2017-11-01 by raptordemon