Protonerd/FX-SaberOS #12: Config menu options increments every time when cycling through

Sligs78, 2017-10-30:
Every time I go through the config menu it increments the options when cycling through the menu.

Not only that, I can only end up at green colors when doing the gravity-color options. I see the tip changing colors, but it's only the color green that gets used. Maybe it has to do with the stuck increment menu cycling.

jbkuma, 2017-10-30:
What hardware are you using and are up upgrading from an older version of
LSOS?  You may need to run the calibration sketch to reset your EEPROM.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Sligs78, 2017-10-30:
DIYino V1, and yes I'm upgrading from an older version of LSOS.

I'll report back later after doing the calibration.

Protonerd, 2017-10-30:
Did you configure the board type correctly? Now we have support for Prime (the one probably you have) and Stardust. There are differences, for example in the aux switch pin assignment. If you use the wrong board type chances are one of the switch used for another purpose cause your problems. Also you need to specify if you are using one or 2 buttons. The software reacts differently depending on whether you have an aux button or not.
Regarding the gravity, I cannot confirm, it works the same way as in LSOS 1.5 If does not, I think the problem is with the button assignment.

raptordemon, 2017-10-30:
how does the button press work for gravity colour with 2 buttons. i had a similar issue that i can't get blue. green and red work ok

Protonerd, 2017-10-30:
Gravity color mix uses the 3 axes of the MPU, which gives you exactly 6 different orientation taking the negative axis direction into account.
Saber tip pointing up: Green+
Saber Tip pointing down: Green-
Saber in vertical position with board/MPU facing up: Blue+
Same with MPU facing down: Blue-
Same with MPU facing to one side (try out): Red+
The other side: Red-

I will make a few drawings to better illustrate the directions.

Sligs78, 2017-10-31:
I calibrated it, but it's still incrementing the options when cycling through the menu.

I just read the gravity color directions from your post, and I finally got the red+ color to show up, but it's hard to get it consistantly; I have to shake the lightsaber a bit. I can't get blue+ or green- to show up.

I'm using the 2 button configuration and the wiring diagram from the DIYino Prime manual. Do I do the wiring diagram from the FS-SaberOS wiki instead?

jbkuma, 2017-10-31:
The diagram in the wiki should be basically the same, just represented a
bit differently for ease of use.  I did add a voltage divider for battery
reading, but I think the rest should be the same.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Protonerd, 2017-11-03:
If you are using 2-button setup, you can toggle/increase/decrease color using the aux button. The main button long press will get you to the next config menu item. If the color change is not consistent, it might be an issue with the MPU calibration. If it's not calibrated correctly, GCM will not work reliably. Reason is simple, it measures along which axis the gravity vector runs parallel to. Gravity is always ~16k in the sensitivity range used. A miscalibrated MPU will wreak havon on this algorithm.

The increments in config might also be a result of button usage. In 2-button mode long press on main will proceed you to next menu. Clicks on main/aux will get you up and down the values. Provided you wired up the right buttons.
Prime:
- main is D12
- aux is D4

Stardust:
- main is D12
- aux is D11

Protonerd, 2017-11-04:
Found the bug to the auto increment of certain config menu items. Try out the new commit in the master branch (tag v1.1).

In the Wiki I made a very good explanation of how Gravity color works, have a look!

closed 2017-11-04 by Protonerd