Protonerd/FX-SaberOS #13: single buttom no sound / config mode selection

shiryu55, 2017-11-02:
1----if I compile using a single button, there is no sound response

#define DIYINO_PRIME
//#define DIYINO_STARDUST

/***** BOARD PINOUT DEFINITIONS ******/

/*
 * Definition of the MAIN and AUX buttons of your saber.
 * These 2 buttons are used to interact with the saber electronics in the hilt
 * With the SINGLEBUTTON compile directive you can configure your saber electronics
 * to use only a single button to interact with your board.
 */

// If your saber has only a single button to interact with the electronics, uncomment the next line
// in case you have 2 buttons (referred to as main and aux buttons) leave this line commented out

#define SINGLEBUTTON
--------------------------++++

I cant enter the programming mode to modify the sensitivity.

I can listen to the following configuration modes, but it does not give an option to sensibility
this is the order in which you list the commands

1 sound font
2 flickerStyle.
3 ightinion style
4 sleep out
5 main color
6 clash color
7 blast color
8 volume

Protonerd, 2017-11-03:
To problem #1: this is a known issue with unknown root cause. Sometimes you change something in the code and sound will simply disappear, even if the change does  not have to do anything with sound triggering. Even just changing the number of neopixels already caused for me this problem. It must have something to do with the compiler but so far I was not able to pin it down...for that I would need an oscilloscope to look at the waveforms to see why the MP3 chips is not responding. Changing the code anywhere else, just adding few dummy lines usually solves the problem.

Protonerd, 2017-11-03:
About the sensibility, you need to include it into the config menu items.

For instance:
enum ConfigModeSubStatesEnum {CS_SOUNDFONT, CS_SLEEPINIT, CS_FLICKERTYPE, CS_VOLUME, CS_LASTMEMBER, CS_POWERONOFFTYPE, CS_SWINGSENSITIVITY, CS_MAINCOLOR, CS_CLASHCOLOR, CS_BLASTCOLOR, CS_BATTERYLEVEL, CS_STORAGEACCESS, CS_UARTMODE}; 

The enum member CS_LASTMEMBER will tell you where the config menu stops, items after it in the list will not be active. This way you can alter how many items you want to see in your config and in which order. I love it!

Protonerd, 2017-11-04:
The "no sound" bug has been fixed. Now with the newest commit you will not have the issue any more. Try it and feed back to me.

Did you understand the explanation how to personalize the config menu?

closed 2017-11-04 by Protonerd

shiryu55, 2017-11-04:
Thank you very much for solving the doubts.
I do not understand well how to do it. if you would put it for clumsy it would be great