Protonerd/FX-SaberOS #14: Couple of issues

kirkyb, 2017-11-02:
2 things...

How does the swing sensitivity option work in the config menu?

A swing sound plays before the end of the saber on audio finishes, do you know why this may be?

Protonerd, 2017-11-03:
The sensitivity option is a bit tricky to explain in short. It goes up to 1g in increments of 1/160 (button click) or 1/16 (hit the hilt). This granularity is due to full range of 16k corresponding to 1g. After changing the sensitivity you can right away try it out, it will play the hum and if the threshold is tripped it will insert a swing.

As to the 2nd item, I've heard it too from time to time, I need to look it up in the code. It could be that the swings are triggered but not suppressed for the duration of the power up.

kirkyb, 2017-11-03:
Thanks for the advice, I'll try hitting the hilt. When I've tried to single press in attempt to change the sensitivity it sends me to the volume setting after a couple of presses.

The swing sound only appears to happen on the first saberon after changing the sound font.

Protonerd, 2017-11-04:
There is a new release as master branch, with some major bug fixes :) I also recalibrated the swing sensitivity, a single click will increase the setting with 1/100th of a g and a hit with 1/10th of a g. You can see the sensitivity increase/decrease by swinging the saber and listening to the swing sound.

kirkyb, 2017-11-04:
Hi Andras,

Thanks for the heads up, do you have a change log?

I gave the swing sensitivity option a go last night using clash, it
certainly worked but at times crashed out or became unresponsive. It'll be
interesting to see how things are in the new release.

I spotted that someone was working on a config GUI too, is this an official
development?

Kirk

jbkuma, 2017-11-04:
Are you referring to the Jakesoft project or something else? Can you share
a link to this project.  I'm not aware of anyone else working on related
projects.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

kirkyb, 2017-11-04:
Yes I'm referring to Jakesoft's STREAM proof of concept, it looks very promising.
It would be great to have a GUI for these boards

Protonerd, 2017-11-05:
Jake's STREAM is for the DIYino boards, but curretly he is busy with other matters, so I guess he will not update it anytime soon. I might ask him if he is willing to give me source code to play around with.

kirkyb, 2017-11-05:
That would be great, having a programme like that would make optimising custom fonts a breeze. From what I can make out, it looks like its using Java?

The evolution of this would be to add wireless capability such as Bluetooth for making quick adjustments, e.g. whilst during a cosplay event etc.

jbkuma, 2017-11-05:
USB or Bluetooth, it's just a serial connection so it's always possible to
add the capability.

closed 2017-11-14 by Protonerd