Protonerd/FX-SaberOS #15: Update v1.1

kirkyb, 2017-11-05:
Feedback regarding the latest update. First of all, well done for changing the sound font option to the first menu slot, this definately makes sense in terms of changing configs quickly and will certainly help to impress users/people. My recommendation would be to put the volume option in the second position as my wife constantly tells me to keep noise down, especially when the kids are asleep.

I'm still having problems with the swing sensitivity option though. Below a certain threshold i don't hear any hum. Just above this threshold i hear distortion, hum starts and rapid successive swing triggers. After a few more clashes, the hum stops, the clashes no longer work but button presses still do. Shortly after this the board restarts and volume is pretty much set to full

Argathena, 2017-11-05:
[image]
To address the Config menu order, you can change this yourself.  I just moved my Battery Level to First, then Sleep Next, and so on.  Just my Preference.  Anyways under the ConfigMenu.h  is where you change the order.  Compare mine to yours and you will see how easy it is to change around.

Protonerd, 2017-11-05:
Thanks Argathena, you said all what had to be said. This setting will be the next one I'm going to write a how-to in the Wiki. Totally crazy, orignally LSOS did not have the sensitivity setting, simply because the setting which was had coded worked so well it was not needed to change it. Then I had a short moment of craziness and implemented a new config item, now many people ask me how it works.

closed 2017-11-14 by Protonerd