Protonerd/FX-SaberOS #16: Sleep Mode Bug

Argathena, 2017-11-05:
After coming out of sleep mode, clash stops working.   Disconnecting Power entirely fixes it.

Protonerd, 2017-11-05:
Cannot confirm. What type of board are you using? I tried 3 different sabers with the new code, all work fine after sleep mode. Is also not logical, sleep mode and clashes have nothing in common.

Argathena, 2017-11-05:
This is on a stardust. Single Button, Star Led

Trying to figure out my other problem with swing detection, i decided to wire up my second Stardust.  Now it wont come out of sleep mode and i can not replicate the bug.  I seem to be having lots of fun, while others are seemless =( Closing since i can not replicate now.

Protonerd, 2017-11-05:
Interesting, I have another saber which also will not wake-up... first I though it has to do with some of the changes I implemented for someone who wished to have staggered ignition for a Kylo Ren saber. Totally interesting, I have no clue at the moment, as my other sabers which all have the same code (mixed Prime and Stardust boards) do not experience the same problem.

I guess not many people use sleep mode, mostly they still use kill keys. Hm, strange, I did not touch the sleep mode code at all and it worked already since LSOS times... I need to have a look at it.

Protonerd, 2017-11-05:
BTW, I had a same saber, single button, Stardust, Star LED. I can wake that one up. I hope it will not develop into a similar problem which kept haunting me where suddenly the sound was gone without any apparent reason.

Protonerd, 2017-11-06:
Today I uploaded the same code again to the saber I could not wake up and now it works...strange. But to be on the same side I added to the sleep entry function again the setting that the main button press wakes up the saber. Try out and feedback to me about how it works for you.

Argathena, 2017-11-06:
Sadly this is not working either =(

Protonerd, 2017-11-07:
What does not work? The saber still does not wake up after going into sleep mode? Hm, it gets stranger and stranger, because you seem to be the only one with the problem...can you send me a wiring picture of your Stardust boards?

Argathena, 2017-11-08:
It wont wake, Still cant replicate the very first "Bug"
[image]
[image]
[image]

Let me know if you need different angled shots

Protonerd, 2017-11-14:
Does the problem with the sleep mode entry still persist? Or has that one got solved with the MPU topic too?

Argathena, 2017-11-14:
It is still not working. I hadn't been working on electronics, as I was making hilt, pics to come. Now that I'm finished I can focus on it a little more. But no, solving the mpu issue did not solve this at the same time. (I was hoping too) Tonight I'll try with my other star dust to see if it works on that one.

Argathena, 2017-11-18:
Working!!! I added an extra led off Ls1 Ls2 and ls3 for my accent light. That way it does what the blade does. Anyways when I soldered that in, It corrected the sleep mode not waking up.  100%

closed 2017-11-18 by Argathena

Argathena, 2017-11-23:
Is the accent light suppose to be on in sleep mode?

Protonerd, 2017-11-23:
Actually not. But I do not have a Stardust saber with an accent LED, only a Prime based saber. So it well might be that if a different pin is used it remains on during sleep mode. From which pin do you drive your accent LED? Is it a single accent LED to drive a button LED?

Argathena, 2017-11-23:
I use a single green led off the A0 pin per the wiki default pin for fx saber os.  Currently it's just an led accent light, not on a button. Since it stayed on all the time, I took out my led button

jbkuma, 2017-11-23:
I think the Stardust pads need to be checked and updated in the wiki, I
based the table off of an earlier pre-release version.

Argathena, 2017-11-25:
Can I reassign led accent pin to A1 or A2 in the code and on board?

jbkuma, 2017-11-25:
On the Stardust those pins are reserved for switching the FTDI and sound
chips off for sleep mode. A0 is the only pin available that has a breakout
unless you use single button, in which case you can use one of those pads.

Argathena, 2017-11-25:
I do use single button. Just trying alternate options to keep sleep mode as optimal as possible. I may also just use a kill key or a switch to save battery. 

Only issue with sleep mode being you have to put it in sleep mode for it to work, and I forget almost every night.

jbkuma, 2017-11-25:
You can add a "last action" timer.  There was such a thing in one of my
earlier versions. Basically you add an unsigned long lastAction and update
it with millis() on button presses and/or activations.

If millis() - lastAction > sleepyTime go to sleep, where sleepyTime is the
duration at which you want it to go to sleep on it's own.

On wake you'll need to set lastAction to millis or it'll just go back to
sleep.