Protonerd/FX-SaberOS #17: Sound is set to max volume everytime when resetting

Sligs78, 2017-11-06:
The sound is set to max volume every time the saber is turned on from completely off. When I get to the volume setting, the volume turns back down to the user settings; but when I reset the saber, the volume is set to max again.

I have CS_VOLUME in front of CS_LASTMEMBER. I changed it from the default CS_LASTMEMBER in front of CS_VOLUME.

Protonerd, 2017-11-06:
I added a piece of code in the setup() routine to set volume to max (31) CS_VOLUME>CS_LASTMEMBER, i.e. if the user do not want to be able to change the volume settings. For my test sabers I removed volume as I want full volume if I swing my saber. But thanks for the hint, I will have a look.

BTW, you can also easily find the place in the code where volume is manipulated, just look for dfplayer.SetVolume. Maybe you find the problem faster than I do. Did you have the problem before this release too?

Sligs78, 2017-11-06:
I didn't have the problem before this release. The only problem I had was the menu increasing increments when cycling through the config menu.

I also found another problem with the volume. Everything is saved at their last setting except the volume. When I return to the config menu and get to the volume setting, the volume is incremented twice.

I tried putting CS_VOLUME after CS_LASTMEMBER while also putting SetVolume to 15 in the code area where it defaults, but it still defaulted to max volume.

Protonerd, 2017-11-06:
Try the new build and let me know it fixes your problem. Volume auto-increment should also be fixed now. It works well on my side.

Sligs78, 2017-11-07:
It works now. All the issues that I brought up are fixed.

Is the light supposed to be off for Ignition Style?

Protonerd, 2017-11-07:
Yes, at the moment there are multiple ignition types only for segmented LED strings. Pixelblades and Star LEDs do not have any, it's a place holder for future development. Want to contribute :o) ?

closed 2017-11-07 by Protonerd