Protonerd/FX-SaberOS #18: Diyino prime

Argathena, 2017-11-11:
After getting my stardust working with fx saber os and a huge user error,  i tried setting it up on my prime.   Now I'm getting no sound. 

I hear feedback when the leds come on, so it's getting some kind of signal.

 I checked the wiring. (Spk1 and spk2 easy) but nothing. I even tried to load Lsos to test.  

OS seems fine. Serial monitor indicates that it is running.  If i had put sounds on wrong I'd still get a sound, and I tried multiple times. I feel like I'm making another easy mistake or I killed my prime. 

Is there test sketch like there is for the stardust? I'm not good enough with code to write it, can only manipulate code:-( 

Please and thank you. 

Ps if it's dead I'm gonna need a new prime :-)

Protonerd, 2017-11-14:
Hi, the test sketch https://github.com/Protonerd/DIYino/blob/master/DIYino_Testsketch_Pro.ino can be used for the Prime as well. Hopefully you have an SD-card already loaded with sound fonts to try. Response from Prime shall be the same, repeating the 1st sound on the SD and calibration of the MPU with measuements logged afterwards.
How did you connect Spk1 and Spk2? To A6 and A7 respecively?

Argathena, 2017-11-14:
Thank you. I kinda thought that sketch would work for both. I'll test. 

Spk1 and spk2 wired to a6 and a7, yes. 

This is me here, so it's prolly simple that's why i needed a test sketch lol

Argathena, 2017-11-20:
Sorry to leave this open, very busy.

Anyways Test Sketch failed and im almost certain its dead.  Since that isnt an issue with Fx-saber im closing.

closed 2017-11-20 by Argathena