Protonerd/FX-SaberOS #19: Homebrew

Himmiko, 2017-11-15:
I make a lightsaber Arduino+mpu+dfplayer.
I connect according to this scheme.
[image]
but nothing works.
that's what turns out:
https://youtu.be/KT2rrpyPJqA

What could it be? all set up according to the instructions.

faeton13, 2017-11-15:
You can uncoment debug functions and check it in arduino through comport
monitor

kirkyb, 2017-11-15:
It sounds like your sound files haven't been copied over in the right order.

Himmiko, 2017-11-15:
this is what the port monitor writes

size of StoreStruct: 72
32
L03
EEPROM LOADED
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful
Initializing DMP...
address: 200
output: -3402

address: 202
output: -278

address: 204
output: 940

address: 206
output: 95

address: 208
output: 9

address: 210
output: 4

Enabling DMP...
Enabling interrupt detection (Arduino external interrupt 0)...
DMP ready! Waiting for first interrupt...
Connecting to DFPlayer...
Connected to DFPlayer !

Himmiko, 2017-11-15:
Sound files are compiled correctly. With the code, JakeSoft works. but there is no Fx.

kirkyb, 2017-11-15:
To me it sounds like the boot, saber on and saber off sounds aren't right.

Himmiko, 2017-11-15:
Yes, the sounds are played differently. but the buttons do not respond ..

Himmiko, 2017-11-15:
Thank you) I figured it all out) reinstalled the program and reinstalled the sounds. Everything is working)

kirkyb, 2017-11-15:
No problem. Good to hear that you sorted it.

Himmiko, 2017-11-16:
how can I help the project? for example, can I send money?))

kirkyb, 2017-11-16:
You could support the developers by purchasing one of the new Stardust boards. Ask Protonerd for more information.

I bought one and it's made making a saber a lot easier. Just make sure that you know exactly what you're doing with it first by thoroughly reading the accompanying manual.

raptordemon, 2017-11-16:
where are these for sale? i couldn't find a buy link

Argathena, 2017-11-16:
https://www.fx-sabers.com/forum/index.php?topic=52302.0

Himmiko, 2017-11-19:
one more question .. what kind of resistors on the battery and button?

jbkuma, 2017-11-19:
For the LED in the button? You'll have to calculate that based on the
button you have, but if it's a 2v 20ma then a 100ohm would do the job.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Himmiko, 2017-11-20:
Excellent) and on the battery? between A3 and +/-

Protonerd, 2017-11-20:
If you supply the circuit from a single LiPo/Li-Ion, you do not need any extra connection, the controller will do the job for you. Only if you use a voltage source >5V and connect to VIN or use a DC/DC on 5V you will need a resistor divider to A3.

Himmiko, 2017-11-21:
I understood everything) thanks)

Himmiko, 2017-11-21:
Well .. the circuit does not work with 
A3 -> +/- 
rst -> +/-
But with Gnd and without A3, everything works fine)

Protonerd, 2017-11-21:
Can you make a sketch of your whole wiring?

jbkuma, 2017-11-21:
If you are using A3 to read the battery you need resistors and indicated to
provide a voltage divider.  Simply connecting the +/- just shorts the
battery...

RST is reset.. there shouldn't be anything connected to this.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Himmiko, 2017-11-21:
jbkuma
> IIf you supply the circuit from a single LiPo/Li-Ion, you do not need any extra connection, the controller will do the job for you. Only if you use a voltage source >5V and connect to VIN or use a DC/DC on 5V you will need a resistor divider to A3.

 connect everything exactly as in this diagram.
[image]

Himmiko, 2017-11-21:
except that my battery 18650

closed 2017-11-24 by Himmiko