Protonerd/FX-SaberOS #20: 4.2 battery

Snyperdead, 2017-11-16:
Hello Andras, 

i have a cuestion, i'll construct a homebrew board for the new FX-SaberOS but my battery its a 14500 lipo battery, 4.2v to 5200 mah.

i need to change the resistors conected to the A3 Pin of the arduino nano????

or what i expecting with this particulary setup?????

i'll apreciate your comments about that.

Thanks in advanced 

Best Regards.

jbkuma, 2017-11-17:
The resistors are for reading the battery level. You should also know
they're is no such thing as a 5200mah 14500, only at least a 26650 can have
that capacity. Some cheap brands falsely label and have really low max
current output as well. Max for an 18650 is 3500, and 14500 is 900 or so.

Protonerd, 2017-11-17:
jbkuma said it all. Nominal voltage of all LiPo batteries is 3.7V, max voltage is 4.2V (i.e. they are charged using CCCV at 4.2V, so the peak voltage can reach 4.2V, but the nominal is 3.7V. The smaller the battery is, the less the capacity.

As to the question how you can measure the battery voltage, if you do not use any DC/DC step-up regulator, but instead supply your setup from the LoPo only, you do not need any connection, because the Nano can measure its own supply, which is the same in this case as the battery. So you can use the code and hardware as is. If you have a voltage regulator in the way it changes things. So far jbkuma found a solution with a voltage divider, because for an unknown reason the analog inputs cannot measure the battery with any accuracy.

Snyperdead, 2017-11-17:
Thanks Friend's for your respond

think i understand better with this context and explanation 

expect this weekend, finish the conections and tests, i hope worked whitouht problems

cheers

closed 2017-11-17 by Snyperdead