Protonerd/FX-SaberOS #21: Jukebox Mode

Argathena, 2017-11-20:
One day you will be rid of me ;-)

Wont compile with Jukebox mode on.

> too few arguments to function 'void JukeBox_Stroboscope(uint8_t*, cRGB)'

Probably something simple again as it always is with me.  Guidance is appreciated.

Protonerd, 2017-11-20:
In fact it could be that it will not work. To be honest I did not validate the Jukebox mode any more, since due to the storage limitation on Stardust probably no one is going to use it... on Prime you can, if you want to use it I need a bit of time to debug it. Is it urgent? I have a lot of other, more saber-critical improvements on my list to work on first.

Solution to the problem is probably simple though. But as far as I remember I could make it work for LED string sabers, not for neopixels though, and this mode is anyway somehow a nobrainer for Star LED setups.

Argathena, 2017-11-20:
There is no rush, but I am using star led.. 

And yes on a stardust, you are very limited. I removed 2 fonts and add Duel Fates song. And that all fits. I mainly want it for a prime, when I order a new one or two that is. Might be next week I order that.

kirkyb, 2017-11-24:
I also tried to activate the Jukebox a while back and received the same error. 

I'm not really interested in this to play the full song, but the first 30 seconds or so in order to set the scene. It would be perfect if you could double press whilst the saber's off to start the song, followed by a single click to ignite the blade. Song choice could possibly be chosen in the menu?

Argathena, 2017-11-24:
Precisely my thought. I'm fine with 1 soundfont and 1 full song. Or 2 fonts and half a song. Specifically Duel Fates! All about scene setting for sure. We all have an mp3 player ;-)

DRagonenko, 2018-01-11:
Wait and hope that this feature will work

BipeFlyer, 2018-06-06:
I hope this helps until there is an official fix.  

I got it to work with the Stardust 3.  I'm using a STAR_LED, so I commented out JukeBox_Stroboscope();  in FX-Saberos.ino
#ifdef STAR_LED
**//**      JukeBox_Stroboscope();
#endif

I then had to modify Light.cpp as follows.  (Changes in bold.)

#ifdef JUKEBOX
void JukeBox_Stroboscope(cRGB color) {

 uint16_t variation = 0;
 uint16_t temp_variation=0;
 cRGB tempcolor;

 for (uint8_t i=0; i<=SAMPLESIZEAVERAGE-1;i++) {
  temp_variation=temp_variation + constrain(abs(analogRead(SPK1) - analogRead(SPK2)),0,512);
  //Serial.println(abs(analogRead(SPK1) - analogRead(SPK2)));
 }
 variation=temp_variation/SAMPLESIZEAVERAGE;
  // assumption -> variation max 280
  //Serial.print("\t");Serial.println(variation);

  **#ifdef PIXELBLADE**
    for (uint16_t i = 1; i <= variation; i++) {
      pixels.set_crgb_at(i, color);
    }
    tempcolor.r = 0;
    tempcolor.g = 0;
    tempcolor.b = 0; // RGB Value -> Off
    for (uint16_t i = (variation)+1; i <= NUMPIXELS; i++) {
      pixels.set_crgb_at(i, tempcolor);
    }
    pixels.sync(); 
  **#endif** 

}
#endif

closed 2019-10-23 by jbkuma