Protonerd/FX-SaberOS #22: Gravity Colors on STAR LED

MattLeshman, 2017-11-20:
Hi. Thanks for all the great job you did programing. It's amazing all the features of the OS.

I'm using Homebrew Arduino NANO with custom PCB Home made and a Single STAR LED RGB. 

I can make it works almost everything but the Colors. I think Gravity Colors is a great idea for LED Strings or Pixel LED. But in a single RGB Star LED is very difficult to configure and mix colors. I think because of the lack of preview at the top of the Blade. 

How can we do it better?
How can I come back to the old set of colors that LSOS has it?

Please help!

Protonerd, 2017-11-21:
I cannot sanely think of a mode advanced way to blend colors on the fly, nevertheless I can make the option available once more to have clickable, pre-set colors instead of the blending feature. Don't expect to have it before well into 2018 though...

As to the difficulty of using Gravity Color with a Star LED setup, yes and no. If you have a chassis and your board is always facing the same direction, it takes about 2 minutes to know the 6 possible directions. Of course if you want to make a saber for a kid, that's another matter...
BTW, have you looked into the Wiki:
https://github.com/Protonerd/FX-SaberOS/wiki/Gravity-Color

?

DRagonenko, 2017-11-23:
I also don't really understand how to do it.

shiryu55, 2017-11-26:
I have the same opinion

I would like to be able  preset, a type of ignition, color, sound source, for each saber selection.

Protonerd, 2017-11-27:
I will definitely put in on the to-do list for the future, I don't want to commit anything as of yet, but I will make it available sometimes in the future.

shiryu55, 2017-11-27:
Thank you very much, for me the option of being able to choose is important.

jbkuma, 2017-11-27:
Hey guys, this is a great opportunity to learn.  2 years ago I had never
programmed an Arduino and I really learned by programming my own sabers,
then modifying LSOS.  If you figure out how to getting working within the
current framework we can work on incorporating it into the main branch
development.  This is the beauty of open source programming and how I ended
up getting involved in LSOS/FX-SaberOS.

While Andras and I definitely want to make FX-SaberOS great for everyone,
we have our own priorities for features and refinements. Since it's open
source it is really a platform from which you can create your own ideas
rather than a closed, unchangeable piece of software.  The versions I use
in my sabers are very customized and sometimes I get rid of entire features
to add in my own.  I'll have an example of this that I'll post later this
week on my website and I'll share it to the FXSOS facebook page.
https://www.facebook.com/groups/FXSaberOS/

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

MattLeshman, 2017-11-27:
Great way of thinking. That was the motivation of starting this issue:

Work all along to do the better way to customize our own SO. Many other
people think like me about the colors so... Why not try to make a version
with a standard menu of frequently used colors in Star Wars Universe.

All together, we can figured out!

shiryu55, 2017-11-27:
I do not use Facebook much, but I sign up.
I hope that more Spanish-speaking people join, that we could create a wiki, in Spanish.

no utilizo mucho el facebook, pero me apunto.
Espero que se una mas gente de habla en español, que por cierto podriamos crear un wiki, en español.

MattLeshman, 2017-11-27:
Yo hablo español!
Soy de Argentina.

shiryu55, 2017-11-27:
Perfecto compi, en telegram tambien tenemos un grupo para habla en español. unete.

DRagonenko, 2017-12-08:
has anyone figured out how to make your color selection ?

Protonerd, 2017-12-08:
What do you mean? You have Gravity color, well explained in the Wiki. Otherwise it's in the pipe to include color profiles, of course in line with the open-source mindset I would be delighted if other could jump in with their ideas to implement it.

MattLeshman, 2017-12-23:
Any news on the implementation of the COLOR PROFILES? Anyone try it?

Protonerd, 2017-12-27:
It's not on the priority list at the moment. I'm still very happy with the Gravity color mix, and I feel myself affirmed now that Plecter copied it in its newest ultra high-end CF 9 board. But it will come for sure. Try to make it happen till end of January.

rorzo, 2018-01-14:
https://github.com/Protonerd/FX-SaberOS/issues/44#issuecomment-357536447

MattLeshman, 2018-02-24:
Thanks to all of you!
Now that Color Profile is fully functional I can Close this issue.

closed 2018-02-24 by MattLeshman