Protonerd/FX-SaberOS #23: Advice (not an issue)

Vulkan1, 2017-11-26:
So far all is working which is awesome, though I need to configure sound which will be fun =P.
I wanted your advice or a picture of how you mount your neopixels inside of the blade, I have a 3 pin connector for unplugging the blade. I’m also assuming you run 2 strips of 144 leds/m inside of the blade (wired in parallel and cut off at 120 leds per strip).
I have a saberforge infinity blade if that helps for comparison.

Protonerd, 2017-11-27:
Actually there are saber smiths more skilled in assembling a neopixel blade than myself. jbkuma, who is a co-developer of FX-SaberOS is one of them, and on FX-Sabers Forum you can consult Space Windu. I myself had reasonable results by simply gluing it together back to back with the adhesive tape they come delivered with and then wrap it up - after soldering the signals/power lines - in packing foam (but I would like to really know where I can order such, so far I salvaged them from furnitures I bought), then put it into a blade, but you should use a transwhite blade with an additional diffuser blade inside for best effect.

jbkuma, 2017-11-27:
I posted a video a while ago, it's not very good but you'll get the idea.
I also just use the existing mounting tape.
https://www.youtube.com/watch?v=mm1h3bXQVCc&t=2s

The video shows my original adapters.  My adapters are now up to version
4.  They are more secure and help protect the solder joints.  I am in the
process of moving and I have other things I am working on so I haven't
posted about them yet, but I will soon.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Vulkan1, 2017-11-27:
Thanks guys I’ll get stuck into it.

closed 2017-11-27 by Vulkan1