Protonerd/FX-SaberOS #24: Soundfonts

FatherCool, 2017-12-03:
Hi, I am very very new to arduino and even more to managing sound files through them. How do I load soundfonts onto my saber so that the code knows where to look for them? Do I need to change anything in the code to suit what I have loaded?

Argathena, 2017-12-03:
What kind of setup you got? Homebrew, Diyino Prime or a Diyino Stardust?

FatherCool, 2017-12-03:
Hi Sorry, I am building a home brew with a Star LED as a starter. I'm also not sure how to get the code to compile once I'm happy as when I click to verify the .ino file it can't find the library files.

I am pretty new to this so take it easy. I have only done basic programming on Arduino previously to control servos and the like.

Argathena, 2017-12-03:
I am not many steps ahead of you. I'd say get the code to compile first then work on sounds cause sounds will do you no good with out the code. 

Have a look at the Stardust Pdf or the Prime pdf. In them are instructions on how to load libraries properly as well as sound. Sound is really easy. I assume you have an micro sd break out in your homebrew set up. 

Format sd card(do not just delete, you have to copy files in a specific order, and erasing instead of formatting won't work)
Copy config sound files. Should be 29 of them, and they need to be copied in order.( highlight all 29, click and drag first file to sd card. Should copy that way) 
Next sound fonts. Using the ones that come with Fx-saber os is highly advised til you fully understand. If you just grab the whole folder and copy, it will copy in proper order. 

But really study those pdfs because they aren't far from a homebrew setup as far as compiling and sound fonts. 

It's gonna take some homework and time to get this down.  Best I can help with my self.

Argathena, 2017-12-03:
https://github.com/Protonerd/DIYino?files=1

FatherCool, 2017-12-04:
Thanks for that. Its very helpful.

The pdf didn't contain what I needed for the code to compile but I did find what I needed elsewhere on the arduino site. I have managed to compile it now.

I can't try it out yet as I am waitkng for my nano, DFplayer and MPU-6050 to arrive. I kust wanted to get prepared by looking through the code.

As soon as it does I will get it mocked up on the breadboard.

I just couldn't see in the code where the soundfonts were being called and how it will know where to look for them (file names). I was hoping to install 3 soundfonts  to take my mood. I will of course follow your advice once I get it running.

FatherCool, 2017-12-06:
On another note while I still wait. How do you set up the initial blade colour and when youve set it how does it remember what colour it was last when you turned it off? Or does it lose the last colour setting once off completely (kill switched)?

Protonerd, 2017-12-06:
https://github.com/Protonerd/FX-SaberOS/wiki/Soundfont.h

jbkuma, 2017-12-06:
Any changes to settings are saved to EEPROM and will be used the next time
the saber is powered.  In the current version there is no way to set
initial color settings.

FatherCool, 2017-12-06:
Ah that makes sense. Does it just power up white then on the first run or no LEDs on at all?

Also back to the code. When I load up the .ino file and click verify it still brings up a large number of orange warnings and such at the bottom. Is this normal?

jbkuma, 2017-12-06:
If it complies then it is probably fine.

FatherCool, 2017-12-06:
Well it completed in the latest IDE but failed in 1.6.5 so not sure really. I didn't touch anything I just compiled it as it was!

jbkuma, 2017-12-06:
1.6.5 is too old

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

FatherCool, 2017-12-06:
Fair enough. I am keen to get it breadboarded and see how it goes.

jbkuma, 2017-12-06:
Careful with your connection, I recommend connecting your pixels directly
to your supply (preferably 3.5-4.2v) or battery (connect the grinds) while
breadboarding. Many breadboards have a high impedance and will get hot and
possibly melt or ignite. My original experiments were exciting...

FatherCool, 2017-12-06:
Noted. I intend to wire it as per the saber really and I will be using an RGB Star tri cree for now. I am planning to have a go with neopixels in the new year though once I have the code down.

closed 2018-01-09 by Protonerd