Protonerd/FX-SaberOS #25: sleep

DRagonenko, 2017-12-04:
What needs to change in order to when entering the sleep mode damped button 
Arduino
https://youtu.be/EzRB5EcYQ34
Что нужно изменить для того, чтобы при входе в спящий режим затухла кнопка 
Arduino

Argathena, 2017-12-05:
It's probably easy for a programmer, which is not me.

 On my wife's saber,  I wired her button accent led in parralel with her tri-cre led. Specifically the red, since she wanted pink. (Red is on with pink selected)

On my saber I wired an rgb led in parallel with the tri cre led for the crystal chamber so the chamber is always the same color as blade.  Food for thought til a programmer can answer your question.

jbkuma, 2017-12-05:
Mirroring the blade is obviously the easiest thing.  Otherwise, you can add
a line that turns the pad off to the sleep routine and turns it back on in
the wake routine.  This should be the default for ACCENT_LED. If not, we'll
be sure to implement it in the next update.

Argathena, 2017-12-07:
[image]

For the time being I changed  accentLEDControl(AL_ON); to  accentLEDControl(AL_OFF);  Just before Jukebox mode in the main loop.  You can see where it is in the Screen Shot.  This is Software way to mirror the led.  

I prefer it to be mirrored to blade but if you want it only off during SleepMode, You could maybe add  accentLEDControl(AL_OFF); to the DEEP_SLEEP 

Im not to sure my self where exactly to put it.  Im just scratching the surface of Coding

DRagonenko, 2017-12-08:
[image]
Thanks for the tip,
I put it in sleep mode and now the button goes out when you go to sleep

accentLEDControl(AL_ON);

Protonerd, 2017-12-08:
I have a new commit where all of these and more are included. Use it and give me a feedback!

closed 2019-10-23 by jbkuma