Protonerd/FX-SaberOS #26: RBG crystal chamber

FatherCool, 2017-12-09:
Has anyone used a standard 5mm rgb led to light their crystal chamber to mirror blade colour. I don't really want to run them parellel to the blade leds as they are running 1A per die whereas the 5mm led will only draw around 20mA per die?

How would I hook this up with FXsaberOS?

jbkuma, 2017-12-09:
Parallel is fine. You just need to properly resistor you LEDs independently.

All of your power connections are actually in parallel or series to your
battery one way or another, some paths are just more complicated.

Protonerd, 2017-12-09:
You can trust jbkuma's word on it. I tried, parallel drive will work. Do not forget, the voltage is parellel, the current is not.

FatherCool, 2017-12-09:
True. 

I don't think that it will work in my case however looking at how I'm wiring the chassis as I am making a plug in chassis where the blade LED side of the circuit is left in the hilt. The crystal chamber will be on the removable portion so the crystal led won't have access to the individual blade LED feeds to be wired in parallel to them. I might have to just do a deactivated crystal as the switches will also be on the hilt side so won't be able to turn it on outside the saber, which arguably is more realistic anyway!

shiryu55, 2017-12-10:
I'm also thinking how to create crystal lighting,
put a led rgb, and light the color depending on the main color selection, and look just like efects of the lightsaber.

This would be possible. ??

shiryu55, 2017-12-10:
fathercool , 
you are using a homebrew plate or a purchased for the chassis, I am trying to build a chassis for my graflex, if you want we could interambiar information for the creation of the chassis.

FatherCool, 2017-12-10:
Hi I am designing the chassis as a bespoke unit for my saber. I am a 3D CAD designers by trade. I will get it 3D printed once complete. I don't know if it would be suitable for your build though as mine isn't a Graflex but a custom hilt that I made from machined aluminium and stainless steel exhaust pipe.

shiryu55, 2017-12-10:
Any way, if you share your chassis , to take info, I would be grateful

FatherCool, 2017-12-10:
No problem. I will post some info once it is done. It will have locations built into it to accept and hold the arduino nano v3.0, MPU-6050 and DFPlayer together with a 18650 battery.

shiryu55, 2017-12-10:
ok , I  use a homebrew model.

FatherCool, 2017-12-10:
Yes mine is homebrew also.

shiryu55, 2017-12-10:
https://www.shapeways.com/product/D8ET7SSAD/graflex-2187-light-v2?optionId=60678043&li=marketplace

I would love to create something like that

closed 2019-10-23 by jbkuma