Protonerd/FX-SaberOS #28: Resistors on A3

FatherCool, 2017-12-18:
On the homebrew star. What are the resistors on the nano pin A3 for?

Cheers

DRagonenko, 2017-12-18:
100 Ком -
470 Ком +

FatherCool, 2017-12-18:
Yes I know the values I was asking what they did. Cheers.

jbkuma, 2017-12-18:
It's a voltage divider to read the battery level.

FatherCool, 2017-12-18:
Ah right so why does it read it and how does it present the info to us? Will the code it work without it?

jbkuma, 2017-12-18:
There is a battery check function and menu item for battery level. It will
work with out it.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

FatherCool, 2017-12-18:
Ah right. Would I need to comment anything out in the code if I didn't fit the resistors?

jbkuma, 2017-12-18:
You just won't get an accurate reading, you can organize the menu sequence
to move battery check out of use.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

FatherCool, 2017-12-18:
I don't need a battery reading so could get rid of it really.

closed 2017-12-19 by jbkuma