Protonerd/FX-SaberOS #29: LEDSTRINGS

DRagonenko, 2017-12-18:
#define DIYINO_PRIME
LEDSTRINGS 
does not trigger the LEDs

Protonerd, 2017-12-20:
Do you have an LED string saber at hand? They are quite rare nowadays.

DRagonenko, 2017-12-24:
cap led 5 mm
version LightSaberOS version 1.5 work
and in the version of  FX-SaberOS not working .

Protonerd, 2017-12-27:
To be honest since LED string sabers are becoming rare nowadays I haven't checked that particular branch any more. It's not on top of my prio list, but I will sure to have a look in the coming weeks. Is it urgent?

DRagonenko, 2017-12-27:
no not urgentwait and hope

DRagonenko, 2017-12-27:
no not urgent
wait and hope

Protonerd, 2017-12-29:
It works perfectly on my side, even wo/any modification.

Protonerd, 2018-01-09:
Any comment? Otherwise I will close the issue as solved.

DRagonenko, 2018-01-10:
in some positions not immediately light up but only after hitting

closed 2019-10-23 by jbkuma