Protonerd/FX-SaberOS #30: Mosfets

FatherCool, 2017-12-18:
Ok guys what mosfets are you using? Part codes would be nice! The ones I have are not correct.

DRagonenko, 2017-12-19:
bd139

FatherCool, 2017-12-19:
But thats not a logic level Mosfet its an NPN transistor and only has a max collector current of 1.5A. To run my rgb tri cree with the colours in parellel it needs to be able to handle 3A minimum as I am driving each die at 1A each.

I am now looking at an STP55NF06L which looks to have the right spec for logic level activation of high power circuits.

Max Vds 60V
Max drain current 55A
Vgs(th) min 1v,  max 1.7v,  breakdown 16V.

FatherCool, 2017-12-19:
Actually I am in error as each led has its own mosfet therefore each only needs to take 1A.

DRagonenko, 2017-12-19:
irfz44n

FatherCool, 2017-12-27:
Ok mosfets sorted but how the hell do you guys solder them? In the breadboard they work great. As soon as you solder them into copper strip board they no longer work properly. There is no contamination across copper paths and all solder joints done well and fast. However take them back out amd stick them in the board and works ok. Agghh. I have also managed to kill 5 mosfets that no longer work at all. It seems that even 1 or 2 seconds on the pin with the iron and its toast!!

jbkuma, 2017-12-27:
I'm not sure why you are having problems, maybe you need to use a hotter
tip for shorter time?

FatherCool, 2017-12-27:
I don't think so. I was using about 350 degrees last night but was having trouble with godforsaken lead free solder so that may have been an issue. Today I used leaded low melt which made a perfect joint in about half a second. Very puzzled and obviously nervous of making any further boards.

Protonerd, 2017-12-27:
Are you sure you are not soldering on a powered board/system? Best way to kill mosfet is when poking around them with a metallic object while connected to battery/other power source. If you accidentally short drain and source, it will instant kill them.

FatherCool, 2017-12-27:
Very sure. I have to hold the wires on a battery to connect it. I can't solder with those in my hands! ;)

FatherCool, 2017-12-27:
Actually I wonder if I am leaving a charge across the internal capacitor of the Mosfet after the initial test? Not sure if that would make a difference?

FatherCool, 2018-01-07:
Well I have just tried soldering these again and ended up with a none working set, again. The joins are good the iron isnt on for long and yet the mosfets leak and stay on slightly even when they are supposed to be off. I tested them with a multimeter prior to fitting, fine, after fitting, not fine, after removing from board, fine again. I am seriously puzzled. I can only think that it is due to flux contamination or something so just bought some different flux. I will try again tomorrow and report back. This is frustrating as the saber works great apart from this!! Maybe I need to brush up on my force powers!

FatherCool, 2018-01-18:
Turned out to be the flux. changed to a different brand and all is well.

ec-cts, 2018-02-21:
Finally, what mosfet have you used that do not give you problems?

Regards,

FatherCool, 2018-02-21:
STP55NF06L

The flux was the issue not the Mosfet. These work great.

ec-cts, 2018-02-21:
Thank you FatherCool.

ec-cts, 2018-02-21:
One more question:

Do you use resistance between G-S of the mosfet?
If so, the value of the resistance is 1.5 koh?

FatherCool, 2018-02-21:
Yes you need resistance there to get the mosfet to release.

The value isn't so important. I think I used a 10k as its what I had.

ec-cts, 2018-02-21:
Again thank you very much