Protonerd/FX-SaberOS #31: Sounds keep hanging - Incomplete play

MycaelRaustin, 2017-12-20:
Hi, I've set-up a homebrew arduino nano with an RGB star.
I followed all the wiring and configurations; 
My saber is powering up and displaying its lights well, but I noticed that sound effects keeps
hanging or keeps getting interrupted. Sometimes it plays perfectly, sometimes it don't.

I've reviewed all my wirings and reuploaded the ino file on my arduino twice-thrice.
Even did a full reformat of my SDcard and followed the copy-paste instructions.

Question:
1.) Could this be related to the calibration or configuration of my MPU6050?

2.) I noticed with LSOS that there was a 470-Ohm resistor connecting to Rx of the DFPlayer. Does anyone know why this is no longer included in FxSaber's set-up? 
Reference: https://github.com/neskweek/LightSaberOS/blob/master/README/Schematic_std.png

3.) What power supply do you guys feed your sabers? I initially tried a vape battery with 3.7V(18650). Right now, I'm using a 5vdc power supply. 

Any help would be great! Keep up the good work!! XD

jbkuma, 2017-12-20:
> README/Schematic_std.png

A 3.7v battery capable of delivering 10a, preferably protected, is
recommended.

The resistor is still recommended for the RX line.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

MycaelRaustin, 2017-12-20:
Thanks for the response, jbkuma :)

Got it. 
Just curious, what's the significance of the 470-Ohm resistor at the RX line?
Do you think having this could help fix the sound-play interruption?

Protonerd, 2017-12-20:
If you supply your board with 5V (i.e. form USB for instance) you will need the resistor for level shifting purpose. If you go for a single 3.7V battery you might omit it.

As to the incomplete plays, your DFPlayer might be defective or your sounds might not be hum-extended. Where did you get your sound fonts?

NATODevTeam, 2018-02-23:
Hi, friends. I nave same problem. Sound fonts I take here https://github.com/Protonerd/FX-SaberOS/tree/master/soundfonts. But after switch off  saber the sound  "powerOff" stil replay and replay non stop. Can you help with this bug?

AirmakZ, 2018-02-23:
Hi NATODevTeam,
Your behavoiur is not the same as the discussed in this topic, but is exactly same as the phenomena I’m facing in my project.
Basically once a track is played in loop mode (like hum) the next single play is executed in loop as well.
It’s not a software bug I think but related maybe with hardware (or hardware firmware).
Cloud you check the device mounted in your DFPlayer? Is it an original XY5200?
It’s just a theory right now. Your information can maybe confirm it.
As a fast test, could you perform a power on and fast power off (in hum sound is not player,

AirmakZ, 2018-02-23:
Sorry, previous message is incomplete.

If hum is not player (it’s using loop mode) and you play power off track during power on track playback you will not see this issue.

roycwalton, 2018-02-24:
If I could piggyback on this open issue so I don't create a new one:

I have recently gotten into the scene as well and constructed the homebrew (nano, mpu6050, dfplayer, and neopixel strand) saber.  However, when it boots up - the dfplayer LED will turn on; it will play the hum (once) and then shut off - no further sound even with blade ignition.

I have traced the line of code to FX-SaberOS ino where the dfplayer is initialized:
 /***** DF PLAYER INITIALISATION  *****/
  InitDFPlayer();
  delay(200);
 // according to debug on 3.11.2017, these 2 lines below cause the sporadic disable of sound. For audio tracker they are not strictly needed.
  pinMode(SPK1, INPUT);
 pinMode(SPK2, INPUT);
  SinglePlay_Sound(11);
  delay(20);

Once this code runs; the dfplayer will no longer respond. Other than that - the boot process is completed and ready for action interrupt.

NATODevTeam, 2018-02-26:
Thanks all for answers.
AirmakZ the chip mounted in DFPlayer clear (top cover burned laser, China made).

AirmakZ, 2018-02-26:
I have samples with the reference deleted with laser too and the same issue.
I cannot be sure because I’m waiting for original XY IC’s, but consider it as a possible root cause.

AirmakZ, 2018-03-09:
Hi! I confirmed what I suspected... After exchange in the DFPlayer module the non XY device with an original XY all works like a charm.
Cheers!

JagdNutCrkr, 2018-05-21:
AirmakZ, where did you end up purchasing the DFPlayer Mini with an original YX IC?  I'm having a similar problem with the looping issue.  I tried two different vendors already from Amazon, but the same results.  I've asked for refund both times because the pictures show they have the YX IC, but don't when I receive them.

Thanks in advance.

AirmakZ, 2018-05-21:
As a reference you should confirm is SMD components are having a white square around solder area.
Also Led color is blue instead of red but it cannot be confirmed in the pictures!
Check this link as reference
http://s.aliexpress.com/BveIZVri

JagdNutCrkr, 2018-05-21:
Thanks for the reference AirmakZ!

closed 2019-10-23 by jbkuma