Protonerd/FX-SaberOS #32: Battery Level

FatherCool, 2017-12-21:
I notice that the battery level is monitored. Is it possible to connect say 3 leds green, yellow, red to this to give an external level indicator? If so what pins do I fix the leds to?

jbkuma, 2017-12-21:
It's possible, but it does not exist as a feature.  The current behavior
for RGB star LED type sabers is to fade the green and red according to the
life span. At full, green is full and red is zero, at 50% green and red are
both at 50% (yellow) at low life red is 100% green is 0.  You can search
for the blade meter function and alter this to suit your needs, be aware
this meter is the same that is used for the volume indication.
Alternatively you can write your own battery meter display routine and
change the call in the menu routine from the generalized version to yours.

On my own heavily modified versions of LightsaberOS I've used pixels to
create circle or bar meters.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

FatherCool, 2017-12-21:
Ah right. I see. Well for now I will just get the saber working as is then tinker with the code. I have a bar of coloured leds on my hilt already which would be nice if it actually showed battery level rather than just lit as it currently is/was.

Protonerd, 2017-12-27:
If I were you I would use a neopixel stick instead of the old style bar graph.

FatherCool, 2017-12-27:
Thats a fair comment but my saber design has a series of individual 5mm leds on a box at the base of the hilt. An old school led bar would suit better.

jbkuma, 2017-12-27:
You can always drive it with ws2811 chips and some clever programming. I
use these chips quite a bit, actually.

FatherCool, 2017-12-27:
Ok I will look into it. I am not desperate for this at the moment. Getting it working at all is No 1.

closed 2019-10-23 by jbkuma