Protonerd/FX-SaberOS #33: Stardust/LED-star combo oddities.

NobAkimoto, 2017-12-22:
I recently got a couple DIYino Stardust boards, and have been putting together a couple sparring sabers using a tri-cree RGB LED star as the light source and a main/aux switch combo on D12/D11. I've loaded 3 of the stock soundfonts on the flash storage.

I've been working with FX-SaberOS 1.1 along with the most recent repository version. The 1.1 release when loaded doesn't seem to actually run - there's a lot of noise/static - the "DIYino Lightsaber" boot-up wav half-plays then goes back to noise/static.

The most recent repository version when set to two-button compile, when the board boots it switches automatically to config mode, then sets to sound fonts - only one button works - and it won't actually switch back to action mode out of config mode.

Enabling Single Button mode allows FXOS to boot normally (and play the "DIYino Lightsaber" boot-up wav), and start out in action mode - but when switching to config mode it only allows adjusting the soundfont - I'm not sure if it has to do with the fact that it's not recognizing double clicks - but the end result is that the functionality of the whole thing is rather limited.

Does anyone have any experience with debugging this and where I might want to look to find the culprit?

jabrooke, 2017-12-22:
I'm having a very similar issue with the Stardust -- same behavior with the initial boot sequence with default test sound, and similar behavior with two-buttons.  I also have an issue where the sound starts playing whenever I apply power, and when I do get the blade to light, the sound goes off.

Protonerd, 2017-12-22:
Hi guys, are you sure you tell the code to use a Stardust instead of a Prime? You can do it in the Config_HW.h file
```
//#define DIYINO_PRIME
#define DIYINO_STARDUST
```
The topic is, on the Prime the D11 is a PWM pin and the D4 is the aux pin signal. If you wire up D11 as a button but use the Prime branch, it will force D11 to 0V (switching off LEDs) and this in turn will be interpreted as a long press on the aux pin, getting the saber to Config Mode right away.

To be honest all my 2-button sabers have a Prime, so I cannot try 2-buttons with a Stardust, but the only difference is the D11 pin (LED driver for Prime and aux pin for Stardust).

One-button mode should work fine, the problem you see there comes from not being able to make a proper double click. It needs some excercise. Actually one of the thing I will work on next year will be to eliminate the double click as much as possible.

NobAkimoto, 2017-12-22:
Hey Andras.

I definitely have it on the Stardust branch - so I'm trying to figure out why it's doing the behavior it's stuck on.

Do you know why it might be acting strangely on Version 1.1, but doing okay on the current version?

NobAkimoto, 2017-12-22:
Also is there a parameter for adjusting the period required for a double click to register?

jabrooke, 2017-12-22:
I confirm that I also have it set as stardust.  I’ll start again in the morning and review the code to see if a prime vs stardust check is not working as expected.

NobAkimoto, 2017-12-22:
I can confirm that one button mode works fine - the double-clicks DO have to be really really fast, but it seems part of the problem was that sleep mode's right after soundfonts, so if you're too slow it puts the board to sleep right after.

Protonerd, 2017-12-22:
The parameters governing the button sensitivity required by the OneButton library are located in the Config_SW.h file:
/*
 * BUTTONS PARAMETERS
 ************************************/
#define CLICK        200 //5    // ms you need to press a button to be a click
#define PRESS_ACTION    600 //200  // ms you need to press a button to be a long press, in action mode
#define PRESS_CONFIG    600 //400  // ms you need to press a button to be a long press, in config mode
/************************************/
 Try playing around with the CLICK parameter to adjust the sensitivity of the double clicks.

Also there is an undocumented feature I dubbed hit-and-run or escape sequence, a hit (clash) detected in Config Mode will have you exit config mode (well, mostly, in volume mode it will set the volume to max).

As to why the 2-buttons mode will start in Config: I'm pretty sure it's because somewhere in the code the D11 is forced to 0V. I just need to find out where. Reason is that D11 is a PWM pin and on the Prime it controls a MOSFET gate.

I also do not really know why the 1.1 behaves differently to the master branch. Of course I alwways improve the code but I do not really remember changing anything other than support for the Pixies and implementing the sleep mode timer.

Can say more if I will have the time to look at the code myself.

Argathena, 2017-12-22:
I do not feel it's a wiring issue. I have both prime and stardust, and i could never get the two button to work on the stardust. I just put one button on my saber and called it good. I never mentioned it because I just settled for one button. I checked the manual a million times trying to wire 2 button on. It was definitely wired correctly. Anyways thought I'd share that similarity with you.

Protonerd, 2017-12-22:
Bug found, fix is in work.

Quick fix for those who cannot wait ;o)
comment out all lines which start with:
`digitalWrite(11`

Protonerd, 2017-12-22:
There are 2 such lines in Light.cpp

jabrooke, 2017-12-22:
Thank you for finding it so quickly!  I haven’t even had enough coffee to open the code up yet.

Protonerd, 2017-12-22:
I made a new release, check out the master branch, it works fine on my side.

Protonerd, 2017-12-27:
I guess it works now as I had feedback to deny it :)

closed 2017-12-27 by Protonerd

NobAkimoto, 2017-12-28:
It works great, sorry for the lack of reply.