Protonerd/FX-SaberOS #34: MPU6050 calibration

FatherCool, 2017-12-27:
Ok I uploaded the sketch to the arduino. How do I get it to run? The wiki says press any key but that does nothing as far as I can see and I don't get anything to answer yes to.

MattLeshman, 2017-12-27:
1st! You need to open the Serial Monitor on Tools.
2nd! Sets the right Baud Rate for communications. (If this does not match the value set up in your sketch in Setup, characters will be unreadable). Example: Serial.begin(9600); Some sketches or other applications may use a different Baud Rate.

MattLeshman, 2017-12-27:
That's in theory.
I do all that but can't configure my accelerometer yet. 
I figured out that my problem with Gravity Color was a bad calibration of my MPU6050.
How we can configure the offsets manually?

FatherCool, 2017-12-27:
But what do I send when I open serial monitor. It seems to do nothing regardless of what I send.

MattLeshman, 2017-12-27:
If you configure the right Baud the serial monitor will send you a message like "press any key to continue" and then you press "any key" and the sketch will set up the offsets.
When it's done the Serial Monitor will ask you if you want to save the offsets to the EPROM. 

In other way, if you don't see any legibly message on your serial monitor, try with other Baud.

FatherCool, 2017-12-27:
The communication baud in the sketch says 115200. I set it to that and it still does nothing. Not liking the look of this!!

Protonerd, 2017-12-27:
If you need to recalibrate your MPU, use this one, same as I use to test my DIYino boards:
https://github.com/Protonerd/DIYino/blob/master/DIYino_Testsketch_Pro.ino

Let us know how it performs for you.

FatherCool, 2017-12-27:
Ok I'll have a go with that once its downloaded.

FatherCool, 2017-12-27:
Ok I ran that and it said connecting to DFplayer and then connected to DFPlayer and that was it nothing else.

MattLeshman, 2017-12-27:
Thank you! I will try that one too.

FatherCool, 2017-12-28:
Well I have still had no life from the calibration program. I also uploaded the fxsaberos sketch to see if that did anything and it doesn't. Whether its because the mpu6950 calibration hasnt run though I don't know.

The nano I am using is a cheap copy though with a ch340 serial chip, would that make a difference or is that what others are using?

jbkuma, 2017-12-28:
It shouldn't matter.  The calibration program is pretty simple from the
user side and should just work, I've used it with nano and pro mini from
various vendors as well as the DIYino boards.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

FatherCool, 2017-12-28:
It doesn't run.

I have uploaded the simple led blibk test sketch to check the board and that runs no problem.

I have just put a multimeter across the 3v3 pin and ground and only got 1.84v which is the wire going to the mpu6050. Do you think that that is an issue?

Would you advise that I rebuild with new parts?

jbkuma, 2017-12-28:
It will definitely need the proper power supplied to it.

Does the script prompt you at all?  Something should be showing up in your
serial monitor.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

jbkuma, 2017-12-28:
If the MPU6050 isn't working your serial monitor should look like this:
`Send any character to start sketch.

MPU6050 Calibration Sketch

Your MPU6050 should be placed in horizontal position, with package letters facing up. 
Don't touch it until you see a finish message.

MPU6050 connection failed`

FatherCool, 2017-12-28:
No I don't receive anything through the serial monitor at all. When I ran protonerds other sketch I got a dfplayer connected prompt but still nothing from the mpu6050 part of it at all.

jbkuma, 2017-12-28:
I just uploaded the calibration sketch to nano with nothing attached to it, straight from the package and got those results.  Are you sure you are uploading the calibration sketch to the board properly?

FatherCool, 2017-12-28:
I believe so. It compiles fine and shows the red RX light and then a little TX light before saying upload done.

FatherCool, 2017-12-28:
Will it work on a mega as I have one of those spare to use as a test bed?

FatherCool, 2017-12-28:
Ok update. I have stripped everything from the board to run it naked so to speak. The sketch with nothing attached now runs and talks to the serial monitor correctly. I can only assume that something was causing an issue connected to the board. The bad news is that I have managed to rip off the copper track on the 3v3 pin completely during de-soldering to the board is a dead stick now for purpose in any case.

New board required!!

jbkuma, 2017-12-28:
It seems like something was bad with the board already since the 3v3 line was low, perhaps it was shorted or the regulator was fried. Fortunately Nanos are plentiful and cheap these days!

FatherCool, 2017-12-28:
Yes indeed. I am just frustrated that I now need to wait to get another!!

closed 2018-01-09 by Protonerd