Protonerd/FX-SaberOS #35: single button change colour

bigjonyz, 2017-12-28:
There seems to be a problem with the chnage colour function in the single button config mode. When I try to change colour the blade does not light up at all except at the very tip, and it is usally a dull purple colour. However it can sometimes become a section of full bright red colour. After changing colour, switching the blade on, the colour becomes a really dull dirty looking light bluish colour not something you would like as a saber balde, and it gets stuck in that colour because going back to config mode to change colour does not work. I don't know this is specific to single button mode or not, as I have only tried the single button setup.

MattLeshman, 2017-12-28:
That's quite a bit similar to my current problem with Gravity Color. And I started to think that's a MPU6050 calibration issue. Maybe.

Protonerd, 2018-01-09:
Most probably it is. Or maybe you do not see the "guide light" at the tip because you did not configure your blade length correctly. It's always the last 5 pixels which indicate the color to be added/deducted. But as a rule of thumb pointing with the blade tip upwards always adds green, downwards reduces green. I.e. you need to see a bright/less bright green color at the blade tip. If this is not the case then something is wrong on your end.
The dirty whitish color is the default setting, it's a weak white with all three colors at 1/3 broightness roughly.