Protonerd/FX-SaberOS #36: neopixel

Himmiko, 2017-12-29:
Hello everyone, people!) which neopixels are you using? 
I bought one tape .. but it does not work. 
Requires too much power.

jbkuma, 2017-12-29:
The pixels need about 20 to 40ma each (rated for 60ma @ 5v but less at
3.7v) depending on the color you use.  for 120 pixels x 2 strips that means
your battery needs to be able to deliver 6 to 10 amps.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

NobAkimoto, 2017-12-30:
I'm trying to set up a homebrew Nano + MPU6050 + DFPlayer Mini as a testbed to try out various configurations of neopixel blades so wanted to doublecheck something while we're on this subject - you don't need n-channel mosfets for doing a pixel blade, right? My understanding was that the D13 signal is what controls the settings on the pixels rather than the combination of FETs for LED-stars.

Is this correct?

jbkuma, 2017-12-30:
You don't need them if you are connecting directly to the battery, but the
pixels will constantly draw power even when they aren't illuminated. The
transistor allows the board to manage power to the blade.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Himmiko, 2017-12-31:
Good. I have 2 18650 batteries. 
they hardly give out such quantity of a current....

DRagonenko, 2018-01-03:
2 аккумулятора 18650 в параллельно и на прямую к neopixel и все  будет работать.
они работают от 2.8-5 В 
https://www.youtube.com/watch?v=y5QoS4JS1qM&feature=youtu.be
здесь 100диодов

Himmiko, 2018-01-04:
я,похоже,какие-то не такие пиксели купил. у меня от аккумуляторов не работает. им нужно 10А тока.
http://led23.ru/catalog/lenty-rgb-begushchiy-ogon-178/lenta-spi-2-2500-5v-rgb-5x-5060-360ledx1-2812-w-017738.html

faeton13, 2018-01-04:
> spi-2-2500-5v-rgb-5x-5060-360ledx1-2812-w-017738.html

​Судя паспорту там всего 5А на метр.
Какие симптомы не работающей ленты?
Подключена через транзисторы или напрямую?
В целом лента должна нормально работать и от одного акума. Идем в точку с
продажей расходников для вейперов, там есть батареи с токовыдачей под 30А.

With best regards.
Taras Protsiv
Additive Manufacturing Researcher

Himmiko, 2018-01-04:
Признаки? хмммм...ну во-первых она не горит)) ни от 1,ни от 2х аккумов. подключена напрямую,через транзисторы,через юсби от компа...как я ее только не подключал.. Батареи я уже заказал 10А,3.7в. Какие вы ленты используете? можно ссылочку?

Himmiko, 2018-01-04:
кст..выпайка транзисторов из ленты помогает запустить 4-5 пикселей от 1 батареи.

faeton13, 2018-01-05:
Транзисторов из ленты? О_о на ней ведь нет транзисторов... Может дело таки
не в питании а линии сигнала? Они не должны включаться без него.
Вобще разные использовал ws2812b обычно. Как раз арлайт должен иметь
совместимы чип.
100диодов по умолчанию в меня запускались от любого не севшего акума 18650
они выдают минимум 6а и этого обычно хватает для 100диодов. Косяки
начинаются после переключения цветов и попытках зажечь белый

DRagonenko, 2018-01-11:
в ленте нету транзисторов. там стоят конденсаторы для работы.
от юсб ты тока штук 10 сможешь включить.
для питания нужно пару аккумов в параллели.

Himmiko, 2018-01-11:
окей. придут попробую.

Himmiko, 2018-01-23:
так..вы оказались правы) высокотоковые батареи справляются с этой лентой. все работает. единственная проблема-я не могу поменять цвет(при попытке шевелить прототипом,если происходит малейшее резкое движение,выходит из меню конфигураций). И после выхода из сна не работает звук. при перезагрезке платы-все окей

Himmiko, 2018-01-23:
https://youtu.be/VtarNnmV-Ho

Himmiko, 2018-01-23:
Так..цвет я поменял,все проблемы решены)

closed 2018-01-23 by Himmiko