Protonerd/FX-SaberOS #37: swing sounds

FatherCool, 2017-12-30:
Ok, big leaps forward here. I have managed to get the thing working pretty sweet however I cannot get it to do any swing sounds at all. It does clashes fine and it plays a swing noise in config but no matter how hard to swing it it makes no swings. 

What could be wrong?

Vulkan1, 2017-12-30:
I’ve come across this error as well, but that’s because I set the swing sensitivity way to high. I’ve havent been able to look at it due to being away for some time.

Vulkan1, 2017-12-30:
Note: I did bring the sensitivity back down but it doesn’t capture swings anymore. I am going to test it on a new board but not any time soon.

FatherCool, 2017-12-30:
Thanks I at least know it isn't just me. I was using the default swing sensitivity though.

Maybe if I run a skwtch to clear the EEPROM and rerun everything, would that make a difference?

Vulkan1, 2017-12-30:
I have been wanted to do that but I’m away from my equipment for another month.

FatherCool, 2017-12-30:
Ok so I cleared the EEPROM and re ran the MPU6950 calibration and installed FXSaberOS.

Works a treat! The swing is now working fine (I feel it could be a little more sensitive but I am swinging a breadboard around!)

Pretty happy now!!

jbkuma, 2017-12-30:
I've found it's probably best practice to run an EEPROM clearing sketch
first, especially with cheapo arduino clones.  It seems even if you write
values you can get some junk that spits out what seems like corrupted
values.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

FatherCool, 2017-12-30:
It seems to have worked fine. I would like the swi g a bit more sensitive I think but if I use the sensitivity configure it get too sensitive too fast and ends up playing one swing over the other.

What value would make it a bit more sensitive but not crazy so if I hard coded it instead of 1000?

Protonerd, 2018-01-09:
You can experiment with values between 500 and 1000 and see which works best for you. Swing sensitivity also strongly depends on whether you use the quaternion method to trigger swing or the less complicated acceleration method. With the former you need a wide sweep in addition to acceleration, with latter only acceleration. I.e. even a high sensitivity will not cause a lot of swings with the quaternion based method.

mendymakesstuff, 2018-01-15:
What should I do even if I dont get swing noises during config? I want to try to clear the EEPROM too! I haven't thought about that yet haha. Going to try that now. Thanks @FatherCool and @Protonerd !

FatherCool, 2018-01-15:
No problem. Theres a sketch to clear eeprom on arduino site.

I have my saber now set at 750 rather than 1000 and its much better for me. I have also tweaked the blaster reset to 500ms too as it goes crazy if its on blaster on swing ans set at 200ms. I may still even make it more than 500 but I will see.

mendymakesstuff, 2018-01-15:
Hi @FatherCool, tried resetting eeprom and running the MPU calibration and reinstalled the OS and... It still refuses to work. Ive tested through teapot and I can see that the MPU moves fine. Ive been stuck on this problem for 2 weeks and I still can't figure out why it isn't working. I just got a new Nano to test with and still the same results. I really do not know what to do at this point.

I just ordered a new MPU. Thats my last option. If it still doesnt work, Im going to go crazy. Lol.

Do you still have any suggestions though? I noticed that I dont get swing sounds even when in config. I dont get swing sounds at all. But all the other sounds work just fine.

FatherCool, 2018-01-15:
It may be that the sound file is not being read for some reason. Maybe format the soundcard and recopy the default fonts across.

Protonerd, 2018-01-15:
What FatherCool menioned is a valid point: where did you get your swing sounds? Are they from another source? The MP3 player will simply do nothing if the files are not recognized... apart from that, did you modify the Soundfont.h? Maybe the swing sound reference goes into Nirvana i.e. not addressing an existing sound.

If you want to see in the serial monitor whether swings are recognized, insert a debug message like Serial.println("Swing");

mendymakesstuff, 2018-01-15:
Going to try that out tomorrow. Another day, another challenge!

FatherCool, 2018-01-17:
Ok still never really satisfied with the swing (always seemed hit and miss as to when it would sound) however I have now disabled the quarternion swing method and to me I get more the feel that I want. I have it set back to 1000 sensitivity now and a 500ms swing suppress. I might even suppress it a little more but I am now much happier.

mendymakesstuff, 2018-01-21:
Hi guys! Just wanted to let you all know that I managed to get my swings to get sound already by starting entirely from scratch and using my macbook instead. Odd that it won't work on windows. Thinking about reformatting my windows laptop and going over things again. But thanks for all the help!

Himmiko, 2018-01-30:
me has a similar problem. when I click the layout with the pinned pixels, crashes in the swings, but without pixels, everything is fine

jayden-jw, 2021-05-02:
Wondering if this project already discontinued since no more update for long. Anyway I have the exact problem, the swing  sound will only work right after flashing or reflashing but as soon as I go to config mode to adjust swing sensitivity it will no longer work. The rest of the functions work perfectly even crash sound.

jbkuma, 2021-05-02:
The project isn't discontinued, it's just mostly finalized and we have been working on other projects.   It is considered essentially the final version.  We have some projects to tidy it up and make it compatible with other boards, but it is not currently a priority.

Please make sure you have formatted and copied the files to your SD card properly, and that you have properly configured both the hardware and software settings before compiling.  The default swing sensitivity should work and if you are having difficulty I do not recommend adjusting it until you figure out what is not working correctly.  You may want to run an EEPROM clear sketch to remove any bad data or settings.  I recommend running this before any install that uses EEPROM.