Protonerd/FX-SaberOS #38: compilation problem

Xamiuk, 2017-12-31:
Hi guys!
nice to meet you! thank you for all the work you do !
i have an issue with the compilation
y have a bad error and i dont now what happens
the error is:

LightSaberOS.ino: In function 'void setup()':
LightSaberOS:224: error: 'StoreStruct' was not declared in this scope
LightSaberOS:228: error: 'storage' was not declared in this scope
LightSaberOS:229: error: 'storage' was not declared in this scope
LightSaberOS:463: error: 'storage' was not declared in this scope
LightSaberOS.ino: In function 'void loop()':
LightSaberOS:596: error: 'storage' was not declared in this scope
LightSaberOS:959: error: 'storage' was not declared in this scope
LightSaberOS:1278: error: 'lightOff' was not declared in this scope
LightSaberOS.ino: In function 'bool loadConfig()':
LightSaberOS:1414: error: 'storage' was not declared in this scope
LightSaberOS.ino: In function 'void saveConfig()':
LightSaberOS:1425: error: 'storage' was not declared in this scope
LightSaberOS.ino: In function 'void __vector_7()':
LightSaberOS:1669: error: expected '}' before 'else'
LightSaberOS.ino: At global scope:
LightSaberOS:1690: error: expected unqualified-id before 'else'
LightSaberOS:1707: error: expected declaration before '}' token
'StoreStruct' was not declared in this scope

i followeed all the instructions and i dont now why i can launch the script.
by the way im using and 
arduino NANO
thanks for you time.

Protonerd, 2017-12-31:
It seens you are using LSOS which became obsolete with the launch of Fx-SaberOS. Please use the latest code with all libraries updated.

Xamiuk, 2017-12-31:
thank you Protonerd, which one is the last code?
thanks for you work and support.

Xamiuk, 2018-01-02:
My mistake!
i commented the led output in Config.h 
that was my mistake.
i uncommented the line and works like a charm!
thanks for the support and this issue is closed :)

closed 2018-01-02 by Xamiuk