Protonerd/FX-SaberOS #40: charging port & battery

Xamiuk, 2018-01-08:
Hi friends!
i hope you doing well,

i have some questions for you if you can response to me please.
for the power supply for my hilt i bought a power bank (mobile external power supply), it has a 18650 battery and a usb charger that charge the battery  like this one:
[image]
[image]

my question is this power bank deliver a 5v power supply to arduino from the usb, that is ok to use in the arduino nano? or i need only the 3,7 v from the battery? and where i need to connect this charger on arduino nano?

thanks for your time and work.
i hope i explain well.

Vulkan1, 2018-01-08:
I did exactly the same and Initially it didn’t work, the circuit board inside the charger needs a load, i Found I had to press the charger button every time for the arduino to turn on, and only then it worked at most for a minute. What I did was extracted the 18650, and added a protection circuit board to it so I could recharge it. I also made it that I could swap the batteries, so I would have a spare on me. If you need clarification I may be able to send some pictures.

jbkuma, 2018-01-08:
You will be better off using just 3.7, this type of device typically can't
deliver enough current and your LEDs, which require the most power, will
run more efficiently from 3.7v.  The boost converter wastes some energy,
then you waste more energy to bring it back down.  You can use a boost
converter to send 5v to the arduino, but this is not strictly necessary.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Xamiuk, 2018-01-08:
Thanks for the quick answer! I found a charger module for arduino and I want to know if is the same that you refer in your comment: 
[image]

Could this module work as you comment?
And vulkan can you show me how you connect your battery to the nano please?

Thank both of you for the responses

FatherCool, 2018-01-08:
I used a high current 18650 for use in e cigs (lg hg2). That has inbuilt charge protection. I have it connected directly to my gnd and 5v pins and it works great. I originally had a 3.7v to 5v booster on the input pin with 3.7v going straight to the LEDs (mosfets) but in the end the booster wasnt necessary and the nano ran on 3.7v like the LEds just fine.

Vulkan1, 2018-01-08:
I currently don’t have access to my pc but I will explain it.
Much like you can plug in a  AA battery to a toy, my 18650 isn’t soldered into the arduinos circuits, it can be unbluged to swap out or in case of an emergency. The 18650’s -&+ go to my PCB (protection circuit board). The PCB has a charger so that’s wired up to a female charger plug. The PCB also has a 5v output. That 5v is sent parallel to the arduino’s 5v and ground pins, and to the neopixels 5vin and ground. 
Are you part of the FB group if so I can add a pic there easier.

Vulkan1, 2018-01-08:
Also I’m still working on mine and so far it’s working but not fully tested

FatherCool, 2018-01-08:
Yes mine isn't soldered in either. I made a plug in battery holder for it inside the saber. You need nothing more. Buy a 18650 with built in protection circuit and bin the other thing. I have a seperate battery charger than can charge lipos nimh or whatever. I just charge a few 18650s and I can swap out all day long.

I don't have a FB account but I understand what you are saying I don't need a picture thats fine.

Xamiuk, 2018-01-08:
facebook group? where?

jbkuma, 2018-01-08:
I have used those charger modules.  You'll need to wire it up just as a
charger, the outputs don't supply enough current to power the blade.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Xamiuk, 2018-01-08:
@jbkuma you mean i need to isolate the charger and take a independent power supply from the battery?
i mean charger just for charge the battery and battery to power supply ?

jbkuma, 2018-01-08:
That charger has tabs to connect a load.  Don't use them, wire the arduino
as though there is no charge circuit.  You can either use the charge
circuit externally with a barrel plug and charge port, wire the charge
circuit in with a charge port, or wire it in parallel to the battery and
use a switch to kill power to the saber while charging.  You should always
have the saber, including the arduino, unpowered during charging to ensure
the charger works properly.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Xamiuk, 2018-01-08:
@jbkuma thank you i will try i will tell you latter if i understand
:)

Xamiuk, 2018-01-17:
hey guys!
the charger  works like charm! i did that @jbkuma said me and connect the charger to the battery and put a switch kill. 
thanks a lot to all of you for the support...
see you guys ;)

Snyperdead, 2018-01-18:
Hello friends, 

thats it's and excellent idea, i figured out, but i haved my Doubts and the pieces for that job are coming, Xamiuk you have an a diagram for the conections????

Xamiuk, 2018-01-18:
@Snyperdead i will draw one for you later.
:)

Snyperdead, 2018-01-18:
Thanks Xamiuk 

I think it's more easier, clear some doubts graphically

Xamiuk, 2018-01-20:
Here's the scheme:
[image]
[image]
[image]
If you have any questions don't hesitate to ask I hope this help you 
:)

Himmiko, 2018-01-20:
nice)

Snyperdead, 2018-01-20:
Thanks Xamiuk 

I will work on this as soon as I have the necessary parts ready.

Best Regards!

Himmiko, 2018-01-29:
Hi) I have one small question. Do you have a problem with the sword when using a single battery and pixels? 
I do not correctly display the movements of the sword and sometimes it reboots. 
perhaps this is due to insufficient voltage. What do you think?

Himmiko, 2018-01-29:
battery 18650 (10a, 4.2V), pixels 144 / m. I use 120 pixels

jbkuma, 2018-01-29:
If it is a real 10a battery it should be ok.  high capacity is better.
With low capacity the voltage can drop too low.

I have been using only 18650 3500mah 10a protected batteries for almost 2
years with pixel blades, no problem.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Xamiuk, 2018-01-30:
@Himmiko you need to connect the pixels directly to te battery, that is the way that i solve that issue, 
my pixels going crazy after a couple of minutes on... but last night i changed the + to the battery and it is solved now... if i am wrong somebody tell me please.
:)

jbkuma, 2018-01-30:
The positive should attach directly to the battery. The negative should be
attached to a transistor which is connected to the battery, or directly to
the battery.

Protonerd, 2018-01-30:
:) The DIYino Stardust V2 has an in-built USB charger already. If you want a reliable integrated solution, go for one.  If you keep adding modules to your home brew solution, sooner rather than later you will run out of room in your hilt.

MattLeshman, 2018-01-30:
How much cost DIYino Stardust V2 and how can a buy it and receibe it to Argentina?

Himmiko, 2018-01-30:
Yes, the battery is original. 
Pixels work fine. problem in the accelerometer and the player. they fail.

StrangeD0s, 2019-02-06:
> If it is a real 10a battery it should be ok. high capacity is better. With low capacity the voltage can drop too low. I have been using only 18650 3500mah 10a protected batteries for almost 2 years [...]

I am a bit confused when it comes to 18650 batteries. How do I know if it is a 10A battery? I have googled it and looked up some datasheets but I'm still unsure. I have a couple of Sanyo UR18650A batteries salvaged from a notebook and I wanted to use two in parallel for higher capacity. I guess I will just try and find out how long they can power a pixel blade.

jbkuma, 2019-02-06:
You probably want a protected cell, and if you are using a parallel pair
you will definitely want to add a protection/balancing circuit.  In
addition to the battery being able to output 10A, the circuit would also
need to.  Known vendors include TCSS and Orbtronic in the US.  Rather than
a parallel pair, you can also use a 26650 which would generally be rated at
15A with protection.

A reliable vendor will give various ratings concerning the battery's
capacity, discharge rates, charging rates, etc.  I would stick to cells
based on Sanyo, Panasonic, Sony, and Samsung.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe