Protonerd/FX-SaberOS #41: Clash sensitivity

FatherCool, 2018-01-08:
Ok I have a wotking saber. Yay! But yiu need to give it a fair whack to get a clash. Is there a way to tweak the clash so that its a tad more responsive?

Protonerd, 2018-01-09:
Then you do something wrong or there is an issue with your MPU. In all of my DIYino builds the clash is hyper-sensitive, even moderate taps with a finger can trigger it. Still it has a superb swing-clash distinction.
The only thing which can screw it up is a too high swing sensitivity if the clash is using the poll method instead of the direct interrupt triggering (see latest release notes). Because if a swing is triggered close to a clash event, the enforced time gaps between FX's will make the clash be skipped.

FatherCool, 2018-01-09:
Well it clashes ok if you tap the hilt with your hand but not with the blade unless its a good smack.

It's ok but not optimal.

Protonerd, 2018-01-18:
Is this still an issue or can I close it? Did you make clash work?

FatherCool, 2018-01-18:
It works but a little sensitive than I'd like. you can close if you like.

closed 2019-10-23 by jbkuma