Protonerd/FX-SaberOS #42: Config button issue

FatherCool, 2018-01-09:
Ok after having a day of blissful saber playing a deal breaker issue has now surfaced. The saber was in sleep mode and then woke itself up without being touched. Now it is stuck in a config loop. It boots and then straight away says config mode I can then scroll through the menu but the the config button now does nothing so I am locked in config. I have cleared the eeprom and recalibrated the mpu and re installed the fxsaberos but as soon as it finishes uploading it now goes straight back to boot and then into config mode.

What could be the issue?

FatherCool, 2018-01-09:
Ok I stripped out the switch wires and redid them and it worked fine everything asbit should be for a good play session. However it has now done it again after leaving it on sleep for around 10mins.

It feels like its a sleep issue but I don't know. Its not the wiring again as I made sure that it was solid this time. Any ideas?

FatherCool, 2018-01-09:
I apologise if I am questioning a lot!

leofaure, 2018-01-10:
Hey FatherCool you are a fuel rats on Elite Dangerous!? I lead for a time a federal clan! great and funny work The fuel rats, fly safe commander.o7!

FatherCool, 2018-01-10:
Hey. Thats right I am a fuel rat. Old school one now though (the youngsters have much more time to do it) and spend most of my time eating snickers in the back of the bar at wollheim vision! 😉

As for the saber button I think I may have narrowed it down but still not sure why it would be and why only after sleep. I think that maybe there is some contaminent between pins on one of the plug connectors in the saber where the wires are attached. When I remove the wires and clean it up it runs ok. I will have to see if the issue resurfaces. Shooting fuel limpets at people is much less hassle!

Protonerd, 2018-01-15:
Straight away config mode entry usually happens if the button pins are wrongly defined. In the code the main button is always D12, but there is a difference based on whether you use Stardust or Prime which pin is the aux pin. If you use the wrong definition and the pin you define is pulled down to GND because it has another function on the hardware, you will end up in config mode right away.

FatherCool, 2018-01-15:
I discovered that on a plug inside my saber I had some flux contaminant between the pin for the aux button and one of the led negs. It was causing a very small current leak which eventually manifested in it reading as if the aux button was always pressed. I cleaned up the connector and all is now well thankyou. The saber works great.

closed 2019-10-23 by jbkuma