Protonerd/FX-SaberOS #43: problems with color on the neopixel

lJusticel, 2018-01-11:
Problem when using fxsaberos, the color of the blade does not change the same as on the tip
https://youtu.be/nxUxHd45pIk

Protonerd, 2018-01-15:
It's a bit hard to see clearly on the vid what is wrong. I see that the tip color is changing, and the blade changes color as well. Actually the blade tip only shows which color you will be able to add or subtract from the current one, i.e. if you have a purple blade (red+blue) and turn the blade to show a dull red at the tip (i.e. subtract red), and keep the button pressed, it will slowly turn the blade color from purple to blue, reducing the drive to the red LED.

Protonerd, 2018-01-15:
See also : https://github.com/Protonerd/FX-SaberOS/wiki/Gravity-Color

lJusticel, 2018-01-15:
blue does not turn on in any way, only the tip is glowing

Protonerd, 2018-01-22:
Try out the new version where you can choose between Gravity Color and Color Profiles. If Color Profiles do not work for you, there is something weird with your stripe.

lJusticel, 2018-01-23:
the new code works well, thank you

closed 2019-10-23 by jbkuma