Protonerd/FX-SaberOS #44: COLOR_PROFILE

rorzo, 2018-01-14:
I don't understand the gravity color, so I want to use fixed colors. In buttons.cpp

```
    else if (ConfigModeSubStates==CS_MAINCOLOR) {
      #ifdef GRAVITY_COLOR
        ColorMixing(storage.sndProfile[storage.soundFont].mainColor,modification,MAX_BRIGHTNESS, SaturateColor);
      #else if COLOR_PROFILE

      #endif
```
COLOR_PROFILE section is empty. Later in CS_CLASHCOLOR, CS_BLASTCOLOR there is no check if COLOR_PROFILE

How can I use pre defined colors?

"A long press on the main button (single button mode) or aux button (2-buttons mode) will slowly adjust the brightness of the corresponding color, this can be used for fine tuning the blade color."
There is no code for brightness adjust in long press aux button section in 2-buttons mode.

Protonerd, 2018-01-14:
Actually in 2-button mode Gravity Color works as well. And yes, currently there is a place holder for fix color profiles, but it's not imlemented yet. It existed in an earlier version of LSOS though. Later I removed because Gravity Color is so much more advanced. Nevertheless I will put it back, but it's not on my prio list. I'm still thinking about how to go about it without breaking the integrity of the code.

rorzo, 2018-01-14:
[Buttons.zip](https://github.com/Protonerd/FX-SaberOS/files/1629725/Buttons.zip)
I changed in buttons.cpp
```
      #ifdef GRAVITY_COLOR
        ColorMixing(storage.sndProfile[storage.soundFont].mainColor,modification,MAX_BRIGHTNESS, SaturateColor);
      #else if COLOR_PROFILE

		}
      #endif
```
to
```
else if (ConfigModeSubStates==CS_MAINCOLOR) {
      #ifdef GRAVITY_COLOR
        ColorMixing(storage.sndProfile[storage.soundFont].mainColor,modification,MAX_BRIGHTNESS, SaturateColor);
      #else if COLOR_PROFILE
		if (ButtonActionType==SINGLE_CLICK){
			confParseValue(modification, 0, 14, incrementSign);
			modification = value;
			getColorFix(modification);
		}
      #endif
      storage.sndProfile[storage.soundFont].mainColor.r=currentColor.r;
      storage.sndProfile[storage.soundFont].mainColor.g=currentColor.g;
      storage.sndProfile[storage.soundFont].mainColor.b=currentColor.b;
      lightOn(ledPins, -1, currentColor, NUMPIXELS/2, NUMPIXELS-6);
      delay(50);
    
      }
    else if (ConfigModeSubStates==CS_CLASHCOLOR) {
	  #ifdef GRAVITY_COLOR
		ColorMixing(storage.sndProfile[storage.soundFont].clashColor,modification,MAX_BRIGHTNESS, SaturateColor);
      #else if COLOR_PROFILE
		if (ButtonActionType==SINGLE_CLICK){
			confParseValue(modification, 0, 14, incrementSign);
			modification = value;
			getColorFix(modification);
		}
      #endif	
      storage.sndProfile[storage.soundFont].clashColor.r=currentColor.r;
      storage.sndProfile[storage.soundFont].clashColor.g=currentColor.g;
      storage.sndProfile[storage.soundFont].clashColor.b=currentColor.b;
      lightOn(ledPins, -1, currentColor, 1, NUMPIXELS/2-1);
      delay(50);
    }
    else if (ConfigModeSubStates==CS_BLASTCOLOR) {
      #ifdef GRAVITY_COLOR
	    ColorMixing(storage.sndProfile[storage.soundFont].blasterboltColor,modification,MAX_BRIGHTNESS, SaturateColor);
	  #else if COLOR_PROFILE
		if (ButtonActionType==SINGLE_CLICK){
			confParseValue(modification, 0, 14, incrementSign);
			modification = value;
			getColorFix(modification);
		}
      #endif
      storage.sndProfile[storage.soundFont].blasterboltColor.r=currentColor.r;
      storage.sndProfile[storage.soundFont].blasterboltColor.g=currentColor.g;
      storage.sndProfile[storage.soundFont].blasterboltColor.b=currentColor.b;
      lightOn(ledPins, -1, currentColor, NUMPIXELS*3/4-5, NUMPIXELS*3/4);
      delay(50);
    }
```
and add function to change color from LSOS 1.4
```
#ifdef COLOR_PROFILE
void getColorFix(uint8_t colorID) {
  switch (colorID) {
  case 0:
//Red
    currentColor.r = 200;
    currentColor.g = 0;
    currentColor.b = 0;
    break;
  case 1:
//Yellow
    currentColor.r = 200;
    currentColor.g = 200;
    currentColor.b = 0;
    break;
  case 2:
//Green
    currentColor.r = 0;
    currentColor.g = 200;
    currentColor.b = 0;
    break;
  case 3:
//Aqua
    currentColor.r = 0;
    currentColor.g = 200;
    currentColor.b = 200;
    break;
  case 4:
//Blue
    currentColor.r = 0;
    currentColor.g = 0;
    currentColor.b = 200;
    break;
  case 5:
//Fuschia
    currentColor.r = 200;
    currentColor.g = 0;
    currentColor.b = 200;
    break;
  case 6:
//DarkGrey
    currentColor.r = 150;
    currentColor.g = 150;
    currentColor.b = 150;
    break;
  case 7:
//DarkOrange
    currentColor.r = 200;
    currentColor.g = 102;
    currentColor.b = 0;
    break;
  case 8:
//DarkViolet
    currentColor.r = 116;
    currentColor.g = 0;
    currentColor.b = 166;
    break;
  case 9:
//DodgerBlue
    currentColor.r = 24;
    currentColor.g = 112;
    currentColor.b = 200;
    break;
  case 10:
//Gold
    currentColor.r = 200;
    currentColor.g = 168;
    currentColor.b = 0;
    break;
  case 11:
//GoldenRod
    currentColor.r = 170;
    currentColor.g = 130;
    currentColor.b = 24;
    break;
  case 12:
//Indigo
    currentColor.r = 116;
    currentColor.g = 0;
    currentColor.b = 204;
    break;
  case 13:
//LightGreen
    currentColor.r = 112;
    currentColor.g = 186;
    currentColor.b = 112;
    break;

  default:
// White (if enough voltage)
    currentColor.r = 200;
    currentColor.g = 200;
    currentColor.b = 200;
    break;
  }
} //getColorFix
#endif
```

Protonerd, 2018-01-15:
If this works, I'm happy! The reason I did not simply want to copy the LSOS code over was that this one uses a colorID to store in EEPROM, and the current FX-SaberOS use a cRGB variable, which consists of 3 bytes. Therefore they are not really compatible. I need to think about how to rewrite the code so that we can still store the same data type as with Gravity Color.

rorzo, 2018-01-15:
It's no need to save ColorId.
At first time color displays from stored value cRGB - then if you want to change color, ColorId saves in local variable "modification" - i checked it used only in GravityColor to define axis, i.e. not used in Color_profile.
Colors save as cRGB - I don't changed it.
Thank you very much for the FX-SaberOS.

Protonerd, 2018-01-18:
If it really works like that I will consider to integrate it inside the code, thanks for trying it out! Really happy that there are still guys out there not only taking the code as granted but willing to tweak it!

Snyperdead, 2018-01-18:
Hellow Friends, 

i checked the code of "rorzo" and working fine in the previous version of fx-saberOS and in the new one,
in the past i try to tweak it, but my knowlege are limited jejeje

thanks for your work folks.

DonSaber, 2018-01-19:
I'm also having hard times trying to configure the color of the blade with Gravity Color on an RGB Star setup. Without the help given by the last pixels in the blade, I find very hard to find proper positions of the hilt to edit the configuration of the three LEDs.
Will try to test Rorzo' solution tonight, if time permits before going to sparring class.

Protonerd, 2018-01-20:
I found rorzo's solution so elegant (simply bringing back an old solution, integrating in seamlessly into a much later version) that I committed it now. Have fun!

Himmiko, 2018-01-23:
Hello) when trying to download the code with Buttons.zip  
A compilation error occurs.

Himmiko, 2018-01-23:
Arduino: 1.8.5 (Windows 8.1), Плата:"Arduino Nano, ATmega328P"

In file included from C:\Users\РћР»РµРі\Desktop\FX-SaberOS\FX-SaberOS.ino:22:0:

sketch\Config_HW.h:46:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\Config_HW.h:89:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST 

       ^

In file included from C:\Users\РћР»РµРі\Desktop\FX-SaberOS\FX-SaberOS.ino:22:0:

sketch\Config_HW.h:109:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\Config_HW.h:154:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\Config_HW.h:173:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

In file included from C:\Users\РћР»РµРі\Desktop\FX-SaberOS\FX-SaberOS.ino:26:0:

sketch\SoundFont.h:23:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

sketch\SoundFont.h:96:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\SoundFont.h:135:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST // 18 files per sound font

         ^

C:\Users\РћР»РµРі\Desktop\FX-SaberOS\FX-SaberOS.ino:111:9: warning: extra tokens at end of #else directive

C:\Users\РћР»РµРі\Desktop\FX-SaberOS\FX-SaberOS.ino: In function 'void loop()':

C:\Users\РћР»РµРі\Desktop\FX-SaberOS\FX-SaberOS.ino:984:26: warning: invalid conversion from 'int' to 'ConfigModeSubStatesEnum' [-fpermissive]

C:\Users\РћР»РµРі\Desktop\FX-SaberOS\FX-SaberOS.ino: In function 'void FX_Clash()':

C:\Users\РћР»РµРі\Desktop\FX-SaberOS\FX-SaberOS.ino:1353:50: warning: integer overflow in expression [-Woverflow]

sketch\Light.cpp.o (symbol from plugin): In function `lastAccent':

(.text+0x0): multiple definition of `getColorFix(unsigned char)'

sketch\Buttons.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Несколько библиотек найдено для "Wire.h"
 Используется: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire
Не используется: C:\Program Files (x86)\Arduino\libraries\Wire
Несколько библиотек найдено для "DFPlayerSerial.h"
 Используется: C:\Program Files (x86)\Arduino\libraries\DFPlayer_LSOS
Не используется: C:\Program Files (x86)\Arduino\libraries\DFPlayer
Не используется: C:\Program Files (x86)\Arduino\libraries\DFPlayer
Не используется: C:\Program Files (x86)\Arduino\libraries\DFPlayer
Не используется: C:\Program Files (x86)\Arduino\libraries\DFPlayer
exit status 1
Ошибка компиляции для платы Arduino Nano.

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

MattLeshman, 2018-01-28:
So. The only code I need to replace is Buttons.cpp?

jbkuma, 2018-01-28:
No, you need to replace most of the code.  You can see which files were
changed in the code page, if you click on each file then click on the note
near the top (currently "Re-include color profiles) you'll see what changes
have been made.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

MattLeshman, 2018-01-29:
Thanks! Thanks! That's what I've been waiting for! I will test it soon!

closed 2019-10-23 by jbkuma