Protonerd/FX-SaberOS #45: Neopixel Segment Count

NobAkimoto, 2018-01-14:
Less an issue but more a question - what's the maximum number of segments a DIYino Stardust or DIYino Prime can handle on a neopixel strip?

I know 120 pixels seems to be the recommended top end limit - but when do the really gnarly effects start showing up in the latest fx-os?

I'd like to try strips of a full meter (144 LEDs) so wasn't sure.

Protonerd, 2018-01-15:
Only experimenting can tell, honestly. Feel free to increase it until it fails. We made a lot of improvements lately, including freeing up some dynamic memory, but I'm reluctant to relax the limit I feel comfortable with.

NobAkimoto, 2018-01-16:
It starts getting finicky past 120 - and it basically stops functioning properly at ~130. (I was going on groups of 5). So it looks like the blade length limit to about 120 segments is still there for the most part at least if you're aiming for reliability/robustness.

Protonerd, 2018-01-18:
I guess it is also depending on some other factors. For instance the fire blade effect needs quite a bit of dynamic memory allocated to it. If the fire blade is disabled (which was feasible in LSOS, but I made it a constant part of the neopixel setup), I can well imaging that one could go up to 130+ . When it comes to pixelcount, it's a tradeoff between FX's and count.

shiryu55, 2018-01-23:
test, version of a week ago, with a single button and 120 led, I produced failures, with two buttons, the faults disappear, however, download the latest version and try again.

Himmiko, 2018-01-23:
Hi) 120 pixels is the limit. further the tape simply does not work.

Protonerd, 2018-01-23:
That is more like a problem with your stripe I guess. If the limit is violated, the whole saber will crash.

Drclic, 2018-02-05:
Hi Protonerd.
How exactly will it crash? Nothing functionnal? No boot? I'm trying to make one and for now nothing happens when i press the main button and I do think my neoixel tripe is too long, but i haven't cut it yet as i'm still in early testing phase, could that be the problem?

Protonerd, 2018-02-05:
What counts is the definition of the number of pixels in the code. The physical count does not matter, at least as long as you do not switch them on (reason is, if you have too many pixels, upon switching on your blade the high current draw can cause an undervoltage and a subsequent reboot/crash, if your battery cannot deliver the needed juice).

Crash means, the system runs out of dynamic memory and will produce strange effects, sudden reboots, all kind of erratic behaviours.

Drclic, 2018-02-05:
Ok. 
So this could be the reason why nothing happens when I push the main button. I was using a 9v to 5v converter provided with the Arduino starter kit and couldn’t boot properly but managed to boot with the nano connected to usb. The led string (neopixel) is not cut, so it’s 5m long.

jbkuma, 2018-02-05:
That converter can definitely not handle powering the pixels.

Drclic, 2018-02-05:
But what about the usb connector of the nano? Could it power up the pixels? It was the way the saber booted, even if I followed the diagram but the power was coming from the nano itself, connected from 5v to the +row on the breadboard. 
I couldn’t read the monitor as I was using my MacBook and for some reason it doesn’t see the com port properly. I’ll try again with the pc.

Vulkan1, 2018-02-05:
The nano won’t be able to output enough voltage for the pixels to function correctly, I believe I tried that myself some time ago

jbkuma, 2018-02-05:
you need several amps to power your pixels, the nano can only provide milli
amps, you will need a 10A capable battery to power your saber, and while
your testing you'll need a supply that can give you that much current as
well.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Drclic, 2018-02-05:
Ok. Duly noted. 
I was wondering, would that work with 3 aa or 3 aaa regular batteries ? Like the first sabers replicas?

jbkuma, 2018-02-05:
AAA aren't going to be useful at all for any light source worth using.
Rechargeable AA's have passable capacity, but can only put out a few amps
safely.  Lithium is really the best way to go.  The best 18650 (typical for
sabers) is about the same price as decent quality AA batteries and will
provide you with a much better power source.  If you are set on using AAs,
you really should use it to power anything more than a star type LED with a
single Cree XPE2 (typical for the hobby), a tri or RGB star type LED may be
a bit much for the AAs depending on the cells.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Drclic, 2018-02-05:
Ok. Thanks for all the information. 
I’m not set on anything really, just considering all options. I thought that three 1.5v batteries would be a good option, at least for testing and mostly because I have some at home 😄 but I don’t mind getting 18650 batteries. 
And just so I know, how about an external usb battery, connected with a cut wire, just vcc and gnd? Could that work, at least for testing?

jbkuma, 2018-02-05:
USB battery is typically some combination 3.7v lithium cells that is
converted to 5v through a circuit which wastes energy, and further they are
typically limited to 1-2A.  Just power directly from a battery or power
supply with sufficient output.  You really only need the 3.7v of the
battery, 5v is just waste heat for the pixels (and they can get HOT).  The
pixels are actually best operated around 4v, which is happily what your
lithium battery will output (nominally 3.7v, full charge is 4.2).  The
arduino will run fine if you connect the 3.7v lithium battery to the 5v
pin, you can get a small upconvert circuit, but I have never found this to
be necessary (although others insist on using it).

Batteries are never really their nominal voltage, except instantaneously
during the discharge cycle.  Alkalines will be around 1.6v or so at full,
with a fairly linear discharge curve down to below 1v.  NiMh will hover
between 1.3-1.2v after a rapid drop off from initial charge, and drop off
dramatically at the end of their useful charge.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Drclic, 2018-02-05:
Thanks a lot for all the informations.
I ordered a couple of 18650 batteries, a charger and a holder that I'll later be able to put in the hilt.
I'll try again when I receive them.

Drclic, 2018-02-06:
Hi again. I should receive my batteries tomorrow but it just crossed my mind, do all 18650 batteries provide 10A? I ordered batteries with 3400mah as the schematics mention 2200Mah, but is it the same thing? I'm not used to amps.

jbkuma, 2018-02-06:
They do not all provide 10A.  If there is a protection circuit it may cut
off below the rating of the battery, you should specifically buy a
protected battery with a 10A circuit.  The capacity (mah) doesn't matter as
much, but lower capacity cannot deliver high current (amps).  Also, cheap
cells should be avoided since their ratings are almost always a lie (this
isn't just a difference or rating procedures, it is outright lies).
Trusted sources include The Custom Saber Shop and Orbtronic in the US, I'm
not sure about Europe or elsewhere.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Drclic, 2018-02-16:
Hi guys. 
I tried again now that I have a dc 5v 10a and 18650 batteries.
I managed to boot properly and turn it on. Only thing, I commented the #define GRAVITY_COLOR line in config_SW.h so I could use the color profiles, but when I get to the main color in the config mode, the led strip (ws2812) turns off, not one pixel on and if I push the buttons, I get the sound confirming I'm changing a setting, but the color of the blade stays white.
I did not change the NUMPIXELS in the Config_HW.H is that the problem?

closed 2019-10-23 by jbkuma