Protonerd/FX-SaberOS #46: Lockup effect

FatherCool, 2018-01-15:
Just a minor one here. When selecting the pulsating flicker style I notice that I no longer get the flashy lockup effect just the sound. Works great on the flickery flicker styles. Where in the code does it initiate the effect so that I can gwt it on for the pulsating style also?

Cheers.

Protonerd, 2018-01-18:
I recall I programmed it a different way: in the pulse flicker style the pulsing gets faster in lockup. If even that does not work, I need to have a look what went wrong.

FatherCool, 2018-01-18:
Ah right, well it looks no different to me. I much prefer the flashy interference effect that the flicker ones have.

Protonerd, 2018-01-22:
Then use that flicker style :)

FatherCool, 2018-01-22:
Haha indeed as I have.

I just liked the pulsing blade for the ancient weapon font!! It's no bother. I thought that I would just be able to copy and paste the flashy lock up effect into the part for the pulsing that's all! I just couldn't find where the code was for the flashy bit!

jbkuma, 2018-01-23:
lockup code was my first contribution, I think, or perhaps the pixel ramp
code.. anyway, it looks like Andras has updated it a bit to represent the
blade colors better.

int lockupFlick = random(0,39);
if (lockupFlick < 25) {
analogWrite(LED_RED, (brightness * maincolor.r / rgbFactor)); // RED
analogWrite(LED_GREEN, (brightness * maincolor.g / rgbFactor)); // GREEN
analogWrite(LED_BLUE, (brightness * maincolor.b / rgbFactor)); // BLUE
} else if (lockupFlick < 35) {
analogWrite(LED_RED, (brightness * clashcolor.r / rgbFactor)); // RED
analogWrite(LED_GREEN, (brightness * clashcolor.g / rgbFactor)); // GREEN
analogWrite(LED_BLUE, (brightness * clashcolor.b / rgbFactor)); // BLUE
}
else { // simple white
analogWrite(LED_RED, MAX_BRIGHTNESS); // RED
analogWrite(LED_GREEN, MAX_BRIGHTNESS); // GREEN
analogWrite(LED_BLUE, MAX_BRIGHTNESS); // BLUE
}
An interesting note here: When I originally came up with this I didn't
reallize random(0,39) is actually an random value between 0 and 38! I found
this out when I was having a hard time getting a random function to work
properly.  The first value is the starting value, and the ending value is a
limit, not the last value.
https://www.arduino.cc/reference/en/language/functions/random-numbers/random/

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

closed 2019-10-23 by jbkuma