Protonerd/FX-SaberOS #47: update fx-saber-os

Xamiuk, 2018-01-17:
hi guys!
me again, just for answer.... yesterday i updated the software on my arduino because i noticed iam using an old version.
first i have a problem with the aux button because it was set on onebutton. i fixed that but when i try to enter to configure mode i dont find the volume configurations and i noticed there is just sound font option, flickering style and color but nothing more...
what im doing wrong?
help me you re my only hope xD
btw i couldn't copy the sd sound files to sd in the correct order in Mac OSX.. i dont know why, i drag and drop, copy paste, cut and paste... nothing work, i change to windows and vòila its work like charm.....  xD

greetings people...
and thanks for the support

jbkuma, 2018-01-17:
Please refer to this article and ensure you have "CS_LASTMEMBER" placed
appropriately.
https://github.com/Protonerd/FX-SaberOS/wiki/FlexiConfig

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Xamiuk, 2018-01-17:
Thank @jbkuma I will check this info
:)

Xamiuk, 2018-01-18:
works like a charm.
thank you @jbkuma

closed 2018-01-18 by Protonerd