Protonerd/FX-SaberOS #48: Creation of a new category in the main menu

shiryu55, 2018-01-23:
It would be possible to add some characters in the main menu system.
i mean if i can as example choose vader and this menu, select color . ignition time ,flicker style, etc.. , preconfigured

Protonerd, 2018-01-23:
Can you describe your idea more clearly?

jbkuma, 2018-01-23:
I believe what Shiryu would like is to have a preconfigured setup where a
sound profile would match a color profile.  For instance: SF1 InVader red
blade, yellow flash.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

shiryu55, 2018-01-24:
correct, jbkmuma, For instance: SF1 InVader red
blade, yellow flash, flicker style, ignition speed, etc.

leofaure, 2018-01-24:
Greetings kyber masters!, i support this idea and maybe I would add a new ignition option, by wrist twist. Ideal for the saber of Kylo Ren!
Thanks to all those who collaborate in this great operating system. We are a few Spanish speakers following from LSOS.
Soon we will upload photos of how our setups are.
Thanks for all.

jbkuma, 2018-01-24:
We welcome and encourage customization to see your dreams come true. Please
feel free to join us on Facebook for discussion and to see other projects.

MattLeshman, 2018-01-30:
I like very much the idea of creating profiles according to characters, linking COLOR with SOUNDFOUNT, and other features for different Characters like Darth Vader, Mace Windu, Luke Skywalker, etc.

jbkuma, 2018-01-30:
To be clear, if there was pre-set profiles it will never be character
names. We've made an effort to avoid copyright and trademarks. This is why
we changed the name from LSOS.

Protonerd, 2018-01-30:
Actually what you ask for already exists, each of the sound fonts have their own customizable profile. You can define it and it stays stored in EEPROM. A profile is always described by a sound font and you can assign all the other profile characteristics to it in the config menu.
If you want to have profiles connected to characters, the easiest way is to replace the fontID sound file to announce the characters name. For instance you replace the fontID of bespin2 (a sound font from GCS) to announce Luke Skywalker. Easy as that.

MattLeshman, 2018-01-30:
Thinking in that way... I think you're totally right!

shiryu55, 2018-01-30:
Thank you very much for the response, it was unsettling me. Looking at the code, my knowledge of programming is a bit outdated.

jbkuma, 2018-01-30:
I believe what we are looking for here is the ability to set the colors in
the font profile such as:

    case 0:
      // soundFont directory 01 DARK INVADER:
      this->powerOnTime = 800;
      this->powerOffTime = 800;
      // Blade Color
      this->mainColor.r = 255;
      this->mainColor.g = 0;
      this->mainColor.b = 0;
      this->clashColor.r = 255;
      this->clashColor.g = 255;
      this->clashColor.b = 0;
      this->blasterboltColor.r = 0;
      this->blasterboltColor.g = 255;
      this->blasterboltColor.B = 255;
      break;

note: This doesn't work, it's just an example of how it COULD work.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

shiryu55, 2018-01-30:
Ok... and could be possible if i can add the fricky style in this profile ??

leofaure, 2018-01-30:
So, can it be understood by my poor translation into Spanish, each sountfont can have different parameters parameters such as brightness, main color or speed of vision, with different values ​​between sountfontos? In that case, could you show the mechanics to follow?

leofaure, 2018-01-30:
It occurred to me that it would be an easy way to use an "eco" mode to save energy or simply by being at night or just to testing , or I set a mode maximum brightness to impress your friends...

jbkuma, 2018-01-31:
At one time I implemented mute more when the hole is powered up while the
activation button is pressed.  I think this could easily be changed to
include a dimmed blade. Search for the quick mute function in the main ino

shiryu55, 2018-02-15:
could you do me a favour answering the question about leo and me, of being able to put predefined configuration parameters in the sound.h. thanks for all.

***So, can it be understood by my poor translation into Spanish, each sountfont can have different parameters such as brightness, main color or speed of ignition , with different values ​​between sountfontos? In that case, could you show the mechanics of the code to follow? Sorry my english,thanks for all, and good work!

jbkuma, 2018-02-15:
This feature does not currently exist.  It is something we are considering
for a future release.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

shiryu55, 2018-02-15:
ok

ec-cts, 2018-02-21:
Hi leofaure.:
I am a Spanish speaker like you. I'm starting my lightsaber project, based on arduino nano, mpu6050 and dfplayer mini.
I'm having a hard time understanding the code and I have some doubts.
My main problem is my poor English, and therefore understand the explanations properly.
There is some way to get in touch with you directly or with PM to ask you some questions in Spanish. I Thank you in advance for the help you can give me.

Greetings to all, you do a great job.

Carlos (ec-cts)

leofaure, 2018-02-21:
Saludos Ec.cts!, te invito a un grupo de telegram que creé hace un tiempo ya, con animo de compartir y colaborar acerca de este gran proyecto creado por grandes y altruistas y genios Protonerd,Jbkuma,y demás colaboradores. Este canal de telegram se creo con la idea de no molestar demasiado al personal de habla inglesa y como canal rápido de comunicación, siéntete libre de saludar y compartir tus dudas e ideas acerca de este proyecto y algunos mas, todos relacionados.

https://t.me/joinchat/AIiqMEKH59010XSDtmAUMQ

Salut!

closed 2019-10-23 by jbkuma