Protonerd/FX-SaberOS #52: MPU6500

MattLeshman, 2018-01-30:
I purchase on eBay from China some MPU6050.
But they send me for error the model MPU6500.

I don't know yet if they will work on FX-SaberOS.

Anyone tried it?
I don't know what to do.

Himmiko, 2018-01-30:
hello!) I ordered my accelerometer here. https://ru.aliexpress.com/item/GY-521-MPU-6050-MPU6050-Module-3-Axis-analog-gyro-sensors-3-Axis-Accelerometer-Module/32555492360.html?spm=a2g0v.search0204.3.2.32ce96b3M4Mltc&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10344_10068_10342_10343_10340_10341_10543_10084_10083_10618_10307_10615_10301_10313_10059_10534_100031_10103_441_10607_10624_442_10606_10623_10622_10621_10620_10142_10125,searchweb201603_28,ppcSwitch_5_ppcChannel&algo_expid=d6913673-cf5a-49a5-8011-2bcccb40f6e0-0&algo_pvid=d6913673-cf5a-49a5-8011-2bcccb40f6e0&priceBeautifyAB=4
it's cheaper

MattLeshman, 2018-01-31:
Thanks for the link.

But I'm trying to NOT purchase accelerometers again. 

I need to know if the MPU6500 will work fine with the project and the code.

If anyone can help me out, I'll thankfully.

If the 6500 won't work. Then I'll buy again the 6050.

Vulkan1, 2018-01-31:
Get a refund, if they have not sent you the correct product inquire for a refund. From what I can read there similar but have different features. The only way to know if it will work is by testing it. Otherwise refund it.

Himmiko, 2018-01-31:
mpu works great with this project