Protonerd/FX-SaberOS #53: Accent LED ON in Sleep Mode

DonSaber, 2018-02-02:
I noticed that when the saber goes into Deep Sleep, the Accent LED is turned ON.
Honestly I don't know if this is a design feature or something I broke while tinkering with the code, but it looks weird to me.
If the saber is supposed to be in Deep Sleep to save power, I expect the Accent LED to stay OFF.
I'm searching in the code since a while, but I can't find where and why the Accent LED is turned ON when the saber goes into Deep Sleep.
Anybody can give me hints ?

Protonerd, 2018-02-02:
How do you wire your accent LED? I have one saber with AL, but the LED is turning off in sleep mode. But it depends on the wiring. If you wire the LED between BAT+ and the AL signal, it will be neg_active, i.e. LED will turn on if AL is turned off (LOW). If you wire your LED between AL signal and GND, it will do it the other way around. I assume that the AL signal is driven actively to LOW before entering sleep mode.

Protonerd, 2018-02-02:
BTW, if you are using a Prime v1, you can actually remove deep sleep mode, it makes sense only with a Prime v1.5 or a Stardust v2. Reason is, even though the controller goes into sleep, all other peripherials will be happily drawing current all the same. Same holds for all home-brew solutions. There it's better to insert the kill key.

DonSaber, 2018-02-02:
My bad, I definitely wired the LED between Bat+ and AL signal.
Probably I did a mistake when drawing my cabling diagram, I'll change the wiring, it's easy enough.
And thank also for the reminder about Deeo Sleep and Prime 1.0.
After testing the Accent will disable it.

closed 2018-02-02 by DonSaber