Protonerd/FX-SaberOS #54: Being sure to understand wiring schematics

Drclic, 2018-02-03:
Hello.
I'm pretty new to all this but I'm willing to get into it.
For now, I'm not sure I get everything right when looking at the schematics for homebrew wiring with neopixel led strip.
For example, I see the Nano A6 connected to both the DF mini and the speaker. How do I exactly do that in real life, two wires starting from the nano, a connection in the middle?

I understand this must be a beginner question, but it would really help, just being sure I do understand it correctly.

FatherCool, 2018-02-03:
It doesn't matter as long as the pins are all connected.

Drclic, 2018-02-03:
That was fast.
Thank you for the answer.
I must have done something wrong then, I tried making a prototype on a breadboard but nothing worked. I'll try again tomorrow.

jbkuma, 2018-02-04:
If you check the serial monitor to see if its booting property.

The purpose of this connection is to read the sound to generate the flicker
effect.

Drclic, 2018-02-04:
Hello.
I tried again, the problem came from the power source (9v to 5v converter that came with the Arduino starter pack). With the Arduino connected to a usb power source, I can hear a boot sound, actually the 011-DIYinoLightsaber_boot_t2s that is located in the configSounds_jbkuma folder.
For now I don't have a led momentary button, just a regular button, nothing happens when I press it.
It is not soldered, just connected with jumper pins.
Do you think this comes from the button or from the sdcard? I'm not 100% sure I did fully understand how the sdcard and sound fonts work, for now I just un-compressed the DIYino_Prime_SDCard_image and sled all folders to the sd card, Selecting them in alphabetical order.
English is not my mother tongue, so when it gets to technical, I can have a hard time understanding exactly what to do.

Xamiuk, 2018-02-05:
Can you show us a image of your wiring or protoboard? Maybe we can help you
:)

Drclic, 2018-02-05:
That's very nice of you.
For now I disconnected everything (to avoid kids and pets to ruin it all) and in other discussions i came to the conclusion that the main problem is my power source: with the converter, nothing works and with the built-in usb connector, can't power-on the blades. So I ordered a couple of 18650 and will try again when I get them. I'll let you know and post a picture of the wiring.
Thank you all very much, I really appreciate you guys are taking time to help.

closed 2019-10-23 by jbkuma