Protonerd/FX-SaberOS #55: Homebrew - Clash Detection and Gravity Blend, but no Swings.

NobAkimoto, 2018-02-04:
Kind of tearing my hair out at the moment with a homebrew that I threw together to test out a configuration for a homebrew combo of an Arduino Nano (V3), MPU6050, and DFPlayer Mini.

I've tried multiple times to recalibrate the MPU6050 using the calibration program - but I can't seem to get it to actually detect swings. It reads clashes just fine, and the direction is also working as I can do gravity blend on it fine. For whatever reason though, it's not acknowledging any movement as swings.

Has anyone else had this problem?

FatherCool, 2018-02-04:
I had a similar problem. I found the swing detection very very insensitive. In the end I changed the swing detection to acceleration only. Maybe yoy should do that and see how it reacts. If ot works you can then decide how you want to move forward as you know that the swing works and its just the detection thats not very sensitive. You could always try lowering the sensitivity value in the code to increase the sensitivity.

Himmiko, 2018-02-04:
I had a similar problem. it was connected to a flash card. format it and load the music files exactly as described in the instructions.

NobAkimoto, 2018-02-04:
Tried that - all the other sounds are working as designed, so I don't think that's a likely contributor.

Himmiko, 2018-02-04:
another question. what type of blade do you use? and what type of food?

closed 2019-10-23 by jbkuma