Protonerd/FX-SaberOS #60: currentcolor issue

whosle, 2018-02-19:
Hello 

 , this is my 2nd attempt at a home-brew version. I am trying the set-up with string blades ( I like bright)  , but will be using all types eventually. 

I am using arduino nano, if it matters. 

The Error I am getting is:

sketch\Light.cpp: In function 'void getColorFix(uint8_t)':

Light.cpp:974: error: 'currentColor' was not declared in this scope

     currentColor.r = MAX_BRIGHTNESS;

Thanks for any help!

Xamiuk, 2018-02-20:
hi whosle, i had this problem before too, check this:

 * BLADE TYPE
 *
 * RGB LED OR PIXELBLADE users:
 * Comment the following line will
 * disable and remove all LEDSTRINGS
 * blocks from compile
 *************************************/
//#define LEDSTRINGS
//#define STAR_LED
**#define PIXELBLADE**
//#define ADF_PIXIE_BLADE

check if this line is uncommented in Config_HW.h the reason why is you define which kind of blade type are you using...

i hope if i help you...

:)

whosle, 2018-02-20:
Hi There Xamiuk,

  Unfortunatly, I have already tried that. 

Whenever the #define LEDSTRINGS is an activated function, the sketch won't compile.

I have successfully compiled it on Star-LED and uploaded my sketch.

I am very happy I have it running, however , I would like the LED String functionality.

Cheers!

Xamiuk, 2018-02-20:
hi whosle, can you show me what homebrew are you using?
the diagram i mean.
:)

whosle, 2018-02-20:
Hi Xamiuk,

 I used the same wiring diagram as provided on here/wiki. The forth /fifth one down in wiring.

 Currently the sketch is running and uploaded, and I have full functionality on all blade types except string blades, which will not compile.

Cheers!

Xamiuk, 2018-02-20:
try something in Config_SW.h
change this configuration:

/* COLOR CHANGE OPTIONS
 *  If you want to use the advanced Gravity Color method developed for FX-SaberOS
 *  leave the GRAVITY_COLOR compile directive defined.
 *  If instead you would like to work with pre-set color profiles (15 different, pre-defined colors)
 *  comment out the line with GRAVITY_COLOR, which will define COLOR_PROFILE to be used.
 *************************************/
//#define GRAVITY_COLOR
//#ifndef GRAVITY_COLOR
  #define COLOR_PROFILE
//#endif

/*
maybe this will help you... :/

Protonerd, 2018-02-20:
Try going back to earlier releases instead of using the master. Since ledstrings became a rarity, I havent checked that blade type lately, so it well might be that it doesnt compile

whosle, 2018-02-20:
Dang.
 Ok,  I will try an earlier release.  I will go back and find the latest release that will work.  

Do you have a guess which release I should use? 

When I track down the issue, I will post my results here.

Thanks again for your effort in this!

Protonerd, 2018-02-20:
no need, I fixed it, was very easy. Try now.

closed 2018-02-20 by Protonerd