Protonerd/FX-SaberOS #65: Occasional mixing up of sound fonts

DonSaber, 2018-03-05:
After bringing the saber to a couple of classes for some "heavy" testing, I have to report a couple of issues that are occurring randomly.
Sometimes it happens that while the saber is on, the fonts suddenly gets mixed up.
I have the feeling that sometimes, for unknown reasons the code is loosing the reference with the calculation of sound files.
I noticed that the hum keeps playing, but the other effects are scrambled, playing the wrong files at swings and clashes.
Honestly I don't know how to troubleshoot this problem or how easy is to replicate it.
I'm running previous FX-SaberOS build with no custom mods, apart from H/W config and MPU calibration values.

jbkuma, 2018-03-05:
Are you using pixel blade, if so how many pixels? If there are too many
pixels there can be memory errors.  Are you using the default fonts?

DonSaber, 2018-03-05:
Sorry, forgot to mention it, I'm using Tri-Cree leds, not Neopixel.
I'm using Plecter fonts coming from Saberfont.com, converted with the Hum Extension tool and reduced to 8 swings and 8 clashes.

Protonerd, 2018-03-05:
I hope its not a memory leak...but if it were one, the saber would go to nirvana after a certain time. I still think it is the wrong file order, but to be honest I never used any of my sabers for half an hour in a row. If I find the time I will also try it...

DonSaber, 2018-03-22:
I wasn't able to test the saber really extensively in the last two weeks, but the problem did not occur again after updating to the latest build.
I'd say we can close the issue for now.
I'll open a new one if I get the issue reproduced again.

closed 2018-03-22 by DonSaber